kliment/Printrun #1177: failed to execute script pronterface RC8

thbl, 2021-03-03:
When i try to open pronterface on windows 10 64x with latest updates i just get "failed to execute script pronterface"

Also tried to build with release_windows.bat but this gives the same error :/

DivingDuck, 2021-03-03:
What language have your os?

Henri-J-Norden, 2021-03-03:
I ended up having the same issue until I managed to build https://github.com/DivingDuck/Printrun/commit/cd8736f550eb55d734810711a093f9af2314bbbc with Python 3.9.2 with wxPython 4.1.2a1.dev5165. (See https://github.com/kliment/Printrun/issues/1174).

@DivingDuck For me at least, that version also gives the 'Failed to execute script pronterface' error.

@thbl If you're into trusting strangers on the internet, you can also try the version I built with the configuration mentioned above: https://norden.io/pronterface.exe

DivingDuck, 2021-03-16:
@Henri-J-Norden, 

https://github.com/kliment/Printrun/issues/1174#issuecomment-791097408

Had you a chance to test this? It should stop "failed to execute script pronterface"

thbl, 2021-04-05:
the version from @Henri-J-Norden  seams to work fine

DivingDuck, 2023-02-12:
@thbl and @Henri-J-Norden 

I know it was long time ago you opened this issue. We got finally a fix for this and other issues.

Here is a link to the latest Windows build.
https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark as completed

closed 2023-02-12 by DivingDuck