kliment/Printrun #1178: Display for Axis positions

waxo62, 2021-03-13:
Hi, I use Pronterface 2.0.0 rc8 for a combo machine 3D-printer, Laser and CNC.  When I use it in CNC and Laser mode I realy miss a simple Axis display on the proterface GUI. It's a lot of jumping up and down to read the positions on the Marlin controlers Smart LCD interface. OK you can do it with G-code M114  but you realy need an all time shown display for X Y and Z. Machine and workpice cordinates can be altered on the same displays if the are positioned under the controller interface down to the left.

Best regards
WAXO

a2k-hanlon, 2021-03-17:
Duplicate of #550, which has been here for a long time but has never been addressed so this should probably be a priority for anyone interested in a new feature contribution.

#706, #965, #986 are also concerned with position reading via M114.

volconst, 2021-03-17:
There are actually 2 sources of coordinates - the coordinates returned from M114, and coordinates that were last requested.
Both sources have pros and cons.
M114 is more real, but adds additional payload.
Requested coords may take some time to become actual.
I would prefer to use requested coords.
To further simplify things, I would not show coordinates during printing, but only while paused and manually jogging.
Do you think after these simplifications the feature is still usable and addresses most of the user needs?

waxo62, 2021-03-17:
Why not use a comination?
Make easy visual displays for X, Y and Z that use requested  coords and calibrate them with M114 every 10 time or so to keep the payload down for Marlin?

Thank to all positive points in this "old" application it will kick som other software far behind for all "Combu users" of 3Dprinting, laser and CNC.

Best regards Sonny

volconst, 2021-03-17:
@waxo62 
Mixing requested and reported coords may frustrate both the user and the coder, Do we need this complexity?
What are you trying to achieve? Don't you just need a feedback while using jog controls.
While manually jogging, requested coords will be almost immediately reached.

waxo62, 2021-03-17:
I just read tho pst from you where you said "M114 is more real, but adds additional payload." If not so, go for requested coordinates......

WAXO

volconst, 2021-03-17:
@kliment , what do you think?

kliment, 2021-03-17:
My opinion on this: There are two situations in which M114 is better - when the device has rebooted and has lost its state (and therefore its last requested position no longer matches) and when we connect to a device mid-operation, and we don't know where it's currently located. So I think we should issue an M114 on connect, and for everything else use requested position only.

kliment, 2021-03-17:
That said, you can already add a custom button to send M114, and this might be a good interim solution.

waxo62, 2021-03-17:
I know to little about the software, but as I see it...
The requested coordinates and the M114 Marlin coordinates should be the same if there are no software lagging. The only error I can see is if these ccordinates don´t match with the actual mechanical coordinates. Then we have a sliping joint or a hardware overload with missing steps. Beacause there are no feedback in the system we can skip this problem and then the most easy solution, requested coordinates or mirrored M114 should be the best way?

WAXO62

kliment, 2021-03-17:
The only time they mismatch is when you connect to a machine without resetting on connect, and the machine is in a particular state but we don't know it and so the position we show doesn't match where the machine thinks it is. That mismatch disappears as soon as we send our first movement command.

volconst, 2021-03-17:
ok, I may try to implement this in april

kliment, 2021-03-17:
@volconst thanks, you're awesome!

waxo62, 2021-03-17:
I´m realy impressed about your response!!!
I wish you good luck with this implententation, and I´m looking forward to test this new function.

Best regards WAXO62

waxo62, 2021-04-25:
Hello to you all again, how does the work flows?
Any positive news about my wishes for the Axis Displays?

Best regards WAXO62

volconst, 2021-04-25:
Hi @waxo62 ,
You can test this branch https://github.com/volconst/Printrun/tree/jog-position
I display the M114 reported positions, not the requested as I originally planned because the M114 were already in place for the jogging.
The coordinates were almost immediately overwritten by an empty string in the status bar, which I disabled.
This may lead to some stale message.