kliment/Printrun #1180: Dialogue box not operating in PF on Mac

Amakin21, 2021-03-17:
Hi! I have my CR10s Pro V2 connected to my Mac running Catalina via usb via pronterface. Not sure which version of PF I am running. I am able to connect to my printer via PF and run commands successfully, but I am not receiving any text feedback in the dialogue box. For example, I just ran a PID autotune via the command M303 E0 S200 C8, which successfully ran my printer through the heat cycles, however I did not receive any text feedback with the values in the box on the far right.

a2k-hanlon, 2021-03-17:
Hi @Amakin21, thanks for opening a new issue. To find out which version of Pronterface you have, please go to Help -> About Printrun in the main menu.

Amakin21, 2021-03-17:
@a2k-hanlon Seems I am running 2.0.0rc5

Amakin21, 2021-03-17:
To be clear, Im supposed to download it by clicking this link only right? or is there something else I need to download with it. https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc8/pronterface-macos-app-2.0.0rc8.zip

if this is correct, then I am now receiving this message when attempting to connect the same way I did with version rc5

Connecting...
Could not connect to /dev/cu.usbserial-AG0JS8D0 at baudrate 115200:
Serial error: [Errno 2] could not open port /dev/cu.usbserial-AG0JS8D0: [Errno 2] No such file or directory: '/dev/cu.usbserial-AG0JS8D0'

a2k-hanlon, 2021-03-17:
Yes, that download should be suffice.

Oh, so you are not able to open a working connection to the printer at all. Could you please open the Terminal application, run this command, and share what is returned? `ls /dev/cu.* /dev/tty.*`

Amakin21, 2021-03-17:
/dev/cu.Bluetooth-Incoming-Port		/dev/tty.Bluetooth-Incoming-Port

a2k-hanlon, 2021-03-17:
Ok. If that is all you get when your printer is powered on and connected to your computer, it means that there is a problem with the physical connection OR you are missing a driver for your printer's USB hardware.

Amakin21, 2021-03-17:
Goodness. If its not one thing its 2 others lol I am mentally exhausted. Thanks for your help man I will try and find a Reality driver set compatible with Catalina

a2k-hanlon, 2021-03-17:
From what I can find online, most of the stock main boards on Creality printers use hardware with this driver for macOS https://www.ftdichip.com/Drivers/VCP/MacOSX/FTDIUSBSerialDriver_v2_4_4.dmg so you could give this a try

a2k-hanlon, 2021-03-17:
However, what I find odd is that you initially said you were able to connect to your printer and send a command that your printer did execute, but you were not receiving any response from your printer. That's despite your computer apparently not recognizing the device and showing a serial port, so I am not really sure what is going on either.

Amakin21, 2021-03-17:
Yeah I was able to send commands to my printer via rc5, I just wasn't getting any feedback in the dialogue box on the right. But now that I've installed the driver via the link you provided. Do I need to do anything more other than clicking the link and following the instructions in the installer to get the drivers installed? Bc after doing so, I ran the terminal command you provided earlier and am still only getting those ports I sent

a2k-hanlon, 2021-03-17:
Well, I'd say try restarting your computer and printer.

Amakin21, 2021-03-17:
I just ran the installer one more time and then restarted, still same ports in terminal :/

a2k-hanlon, 2021-03-17:
Dang. Just curious, can you send commands to your printer with rc8 as you were able to with rc5?

Amakin21, 2021-03-17:
No, its not even connecting at all for some reason. Which is weird bc like you said, I was able to at least connect and send command in rc5

Amakin21, 2021-03-17:
The error message I am receiving now is new to rc8. I wasn't receiving error messages in rc5, I just wasn't able to see the responses from the code commands I enter in the dialogue box. but now with 8, I am unable to connect my printer

a2k-hanlon, 2021-03-17:
Alright. That seems very strange to me. I'm sure it does to you too! The fact that you are not seeing a serial port for your printer at all suggests that the issue is with your setup (printer hardware, connection, drivers on your computer) rather than Printrun, but we need to isolate the issue. In your position the next thing I would try is to connect your printer to a different computer, if you have one at hand. You could also try using a different USB cable I suppose.

volconst, 2021-03-17:
@Amakin21 , when you succeed at connecting with rc5 version, what is the value in Port dropdown list?

a2k-hanlon, 2021-06-19:
Closing due to inactivity. @Amakin21, please feel free to reopen this or open a new issue if you need help in the future.

closed 2021-06-19 by a2k-hanlon