kliment/Printrun #1181: Pronterface Configuration File

djmeltdown, 2021-03-27:
I'm having the same issue  #879 but I can't find the configuration file to change the UI back. Where is it supposed to be? I can't find it in the Printrun-win-18Nov2017 folder that I extracted.

rockstorm101, 2021-03-27:
Hi @djmeltdown,

You are using a very old version of Printrun, I would recommend you get the latest release [[1]]. Otherwise, please see comments [[2]] and [[3]].

> The configuration file on Windows is typically found in " C:\Users\<yourusername>\printrunconf.ini".

> Go to c:\users(yourusername)\AppData and check the subfolders for files named pronsolerc or printrunconf.ini and delete those. If you can't find them search your users folder for those names.

[1]: https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8
[2]: https://github.com/kliment/Printrun/issues/879#issuecomment-359346976
[3]: https://github.com/kliment/Printrun/issues/879#issuecomment-374871735

Duplicate of #879

djmeltdown, 2021-04-03:
Thank you. I've downloaded the zip and run the .exe. I've connected to the printer, but it won't home. It gives me a halted error.

volconst, 2021-04-06:
@djmeltdown Halting during G28 command does not seem to be a pronterface issue.
See https://github.com/MarlinFirmware/Marlin/issues/12952

rockstorm101, 2021-05-02:
Two different problems reported. The one related to finding the configuration file was (successfully) answered. The one related to halting during homing does not seem to be related to Printrun. I'm therefore closing this issue, please feel free to reopen if needed.

closed 2021-05-02 by rockstorm101