kliment/Printrun #1182: p.send_now("M105") and p._readline_nb() issue

comes04, 2021-03-30:
Hi,
I try with this code to get temperature info from my Ender 3 3D printer with the following code:

`from printrun.printcore import printcore`
`import time`
`p = printcore()`
`p.connect('COM4', 115200)`
`while not p.online: time.sleep(0.1)`
`p.send_now("M105")`
`temp = p._readline_nb()`
`print(temp.decode('utf-8'))`

i expect to receive something like this:
`ok T:21.91 /0.00 B:20.62 /0.00 @:0 B@:0`

but instead I just receive some parts of the information but never all of it, for example:
`kT2.8 0.0:0./.0:B:`
or
`k:170.0B2.10.0 :B:`
or
`k216/.0 :02 . :0B@`
etc

Could someone help me with this?
Thanks in advance

volconst, 2021-03-30:
Hi,
you use too low level _readline_nb which is meant to be called asynchronously and is returning chunks, not whole lines. See where it is called.
May be this will work:
while True:
  line = p._readline()
  if line is None:
    break
  else:
    print(line)

volconst, 2021-03-30:
@comes04 ,
First I thought that the problem is that line is read in chucks, but now I see that the format is broken, characters (bits) are missing.
Check your baud rate, it must match whatever the firmware is expecting. If you have custom compiled firmware it may have custom baud rate.
Do Pronterface, Cura work well with the printer? M105, printing, jogging?

volconst, 2021-03-31:
This happens because connect() starts a thread with the _listen method and you read concurrently from 2 threads.
You have to put your line handling in event handler:
```
def temp_callback(line):
  print(line)
p.tempcb = temp_callback
```
and do not read from main thread.

comes04, 2021-04-02:
Hey thanks, this is actually working!!

closed 2021-04-30 by volconst