kliment/Printrun #1184: Getting "ypeError: object of type 'NoneType' has no len()" when trying to open a stl

JamesGordonII, 2021-04-05:
I am trying to print with prontface but when ever I try to open a stl file it gives me this error: ``` [Errno 2] No such file or directory: 'C:\\Users\\Alienware\\Downloads\\LCD_and_RPi_Enclosure_v1_export.gcode'
Traceback (most recent call last):
  File "printrun\pronterface.py", line 1672, in loadviz
  File "printrun\gcview.py", line 349, in addfile_perlayer
  File "printrun\gl\libtatlin\actors.py", line 425, in load_data
  File "printrun\gcoder.py", line 251, in __len__
TypeError: object of type 'NoneType' has no len() ```

is there any workaround for this?

volconst, 2021-04-06:
I think this is because you use a default (software fallback) implementation of OpenGL.
This is common if you run in virtual machine. The workaround is to disable 3D view or install hardware accelerated opengl driver from the video card manufacturer.

rockstorm101, 2021-04-06:
> `[Errno 2] No such file or directory: 'C:\\Users\\Alienware\\Downloads\\LCD_and_RPi_Enclosure_v1_export.gcode'

The facts that an STL file is being loaded and a "No such file or directory" error is being thrown points in the direction of the slicing software not being properly configured. @JamesGordonII what does your option 'Settings > Options > External Commands > Slice command' look like?

JamesGordonII, 2021-04-07:
Hi sorry, I will get back to this when I get home and will be able to use
my pc

JamesGordonII, 2021-04-07:
@rockstorm101 i now realized that it needed to be sliced as you said and doing that and using the gcode it generates and it now works flawlessly

closed 2021-04-07 by JamesGordonII