kliment/Printrun #1185: Pronterface gone back to initial state

highviewstudios, 2021-04-10:
My pronterface has gone back to the initial state after you first installed it. My custom buttons I have made on the bottom of the screen have gone.

Can you tell me why this has happened please? Also, can I somehow get all the button data back please?

DivingDuck, 2021-04-10:
This happen if you swap between v1.6 and 2.0RC version. Search for printrunconf.ini and possibly a backup version *.bak too.

DivingDuck, 2021-04-11:
@highviewstudios 
Had you solve your problem?

highviewstudios, 2021-04-11:
Hi, sorry, yes it did help me so far. I have found the file 'printrunconfig.ini' but it's been replaced by a new one and there's no backup to the file. Thanks anyway for your help :)

closed 2021-04-11 by highviewstudios

DivingDuck, 2021-04-12:
Your welcome.
There are two places you can look in case your OS is windows. The first one is c:\users\YourUsername\printrunconfig.ini and the second one is c:\Users\YourUsername\AppData\Local\Printrun\Printrun\printrunconfig.ini
The second location is within the hidden folder AppData. You need to make it visible.