kliment/Printrun #1186: question about printcore?

lunknowl, 2021-04-13:
hi, this is more question than issues. i try to control 3d printer using printcore using example:

pythonconnect.py
"
from printrun.printcore import printcore
from printrun import gcoder
p=printcore('/dev/ttyUSB0',115200) # or p.printcore('COM3',115200) on Windows
"

pythonhome.py
import pythonconnect
connect.p.sendnow("G28")

my question is: the home program run fine but after that it was stuck in loop, until i disconnect otherwise the python window not exit, like python waiting or stuck in printer connection now. cant execute anything else (home2.py)

volconst, 2021-04-13:
Do you mean that if you have
```
connect.p.sendnow("G28")
connect.p.sendnow("G28")
```
It executes only the first one?
Does the second command get lost for any other combination of 2 commands?
It is by design that the script does not exit until you call p.disconnect() because printcore() constructor connects automatically and connect() starts read, write threads which exit only after disconnect.
If you need to respond to printer replies, handle them in events, see https://github.com/kliment/Printrun/issues/1182#issuecomment-810931984
You may also find this useful https://github.com/volconst/Printrun/blob/master/printrun/gsync.py

lunknowl, 2021-04-14:
well, my problem is, if i use pyserial connect to com7 (my printer com port) when i run (connect.py - connect to com7, only this line "p=pyserial('/dev/ttyUSB0',115200) # or p.printcore('COM3',115200) on Windows") the python run and close the cmd window. but when i run printcore (connect.py - connect to com7, "p=printcore('/dev/ttyUSB0',115200) # or p.printcore('COM3',115200) on Windows")  , the cmd window not close, it waiting for something, and if i put the p.disconnect, it  disconnect before execute the second "G28", i need like a "wait to finish command then do p.disconnect"
connect.p.sendnow("G28")
connect.p.sendnow("G28")

volconst, 2021-04-14:
The easiest way is to put
```
sleep(3 secs)
p.disconnect()
```
The sleep should be enough to send the command to the printer.
The command itself may not finish - marlin may be able to acknowledge commands before they finish, but even if you disconnect from the printer it will probably finish its homing disconnected. You may test it, it may work.
If you need to be certain that the home has executed, you probably need to send a command that is guaranteed to be acknowledged after it is executed (e.g. M114?) and wait for its acknowledge in p.recvcb handler and disconnect from the handler.

closed 2021-06-06 by volconst