kliment/Printrun #1187: printing gcode from python

comes04, 2021-04-20:
Hi there!
One more time i need your help with a proyect;
I'm trying to print a gcode from python, actually a run it trought pycharm in windows.
I just copied the example in this website as following:

`from printrun.printcore import printcore
from printrun import gcoder

# PROCESAR GCODE Y EMPEZAR A IMPRIMIR
p = printcore()
p.connect('COM4', 115200)

gcode = [i.strip() for i in open('prueba_lecturas_pasos_cortos.gcode')]
gcode = gcoder.LightGCode(gcode)
p.startprint(gcode)`

with this, it doesnt print anything.
Any idea what i'm doing wrong?
Thanks in advance.

comes04, 2021-04-21:
Ok, I just need to give time to conect with 
while not p.online:
    time.sleep(0.1)

The following works

`p = printcore()
p.connect('COM4', 115200)

while not p.online:
    time.sleep(0.1)

gcode = [i.strip() for i in open('prueba_lecturas_pasos_cortos.gcode')]
gcode = gcoder.LightGCode(gcode)
p.startprint(gcode)`

volconst, 2021-04-21:
Thanks @comes04 , I updated the README with your fix.

closed 2021-04-30 by volconst