kliment/Printrun #1188: getting coordinates with M114 with printrun

comes04, 2021-04-26:
Hi there,

¿Can someone let me know how to send the command M114 and get the data that returns with printrun?

I know that this is a topic that has been around many timesbut I can't figure out how to get it.

For example, i get the return from command M105 with the following and works perfectly:

```
p.send_now("M105")
def temp_callback(line):
	print(line)
p.tempcb = temp_callback
```

I tried the same with M114 but doesn't work since that command send the info to the serial...

I also have seen the “New module to synchronously, blocking send gcodes with return” which seems to be able to help on this, but i'm not able to use it.

Also tried to send M114 and read it with the  “Serial” library, which works. However i would like to do it just with the printrun library since in order to use “Serial” I have to disconnect “printrun” …

Thanks again for all the help :)

volconst, 2021-04-27:
@comes04 Why you can't use #1008 ?

volconst, 2021-04-27:
You can use:
```
def recv_callback(line):
	print('got', line)
p.recvcb = recv_callback

p.send_now("M114")
```
It receives all responses and you have to which of them is M114.

comes04, 2021-04-29:
Hi volconst,
I tried that and is working great.
Thanks for all your time and help!

closed 2021-04-30 by volconst