kliment/Printrun #1190: printcore m43 gcode

lunknowl, 2021-05-05:
hi, i try to use m43 p63 w to watch the pin 63, the pronterface and pronsole work great but how to i used it in python
def recv_callback(line):
	print('got', line)
p.recvcb = recv_callback

p.send_now("M43 P63 W ")

this is what i have now

volconst, 2021-05-05:
Do you see 'got' lines printed in log/console when pin value changes?
Toggle 'Debug communication' in the menu to see printer responses.
You may need to add
```
while not p.online:
  time.sleep(0.1)

p.send_now("M43 P63 W ")
```

lunknowl, 2021-05-07:
`from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

p = printcore()
p.connect('COM16', 250000)
while not p.online:
    time.sleep(0.1)
    
def recv_callback(line):
    print('got', line)
p.recvcb = recv_callback

p.send_now("M43 P63 W")

p.disconnect()
sys.exit(0)`

this is my code and i got only 1 line
`Got Watch pin`
i try to do this, if pin is high, when i run M43 P63 W, result is high
if pin is low, when i run M43 P63 W, result is low
like a check sensor program

volconst, 2021-05-07:
You should not disconnect and exit if you are waiting for response from printer.
The callback is invoked in a reading thread later when responses arrive.
The other code is invoked in main thread.
You have to disconnect after you have received what you expect, probably calling disconnect in the handler or in a try... except block on keyboard exception.
Do not sys.exit(). Program should exit after disconnect terminates reading thread and the main thread has no more commands.

lunknowl, 2021-05-07:
`got x_min: open

got x2_min: open

got y_min: open

got z_min: TRIGGERED

got ok``

this is what i got but stupid question, how do i get the status of x_min only. i just want to check if x_min is triggered  only thank

volconst, 2021-05-07:
```
ln = 'x_min: open' # test data
if 'x_min:' in ln:
   x_min_open = 'open' in ln
```
or
```
import re
m = re.match('([a-z_]+): (open|closed)', ln)
if m and m.group(1) == 'x_min':
  x_min = m.group(2))
```

lunknowl, 2021-05-07:
`def recv_callback(line):
    print(line)
    ##
        
    
p.recvcb = recv_callback

p.send_now("M119")

ln = line # test data

if 'x_min: TRIGGERED' in ln:
        
    x_min_open = 'TRIGGERED' in ln

    print(x_min_open)

else:
    x_min_open = 'OPEN'

    print(x_min_open)`

i got it working but only when i put under "def" if i move under "p.send.now"  name 'line' is not defined

"ln = 'x_min: open'"     #how can i get data from "def"?

volconst, 2021-05-07:
Replace ln with line.

lunknowl, 2021-05-07:
`def recv_callback(line):
    print(line)
    ##
p.recvcb = recv_callback

p.send_now("M119")

if 'x_min: OPEN' in line:
    x_min_open = 'OPEN' in ln
`
NameError: name 'line' is not defined, can you have me which this?

volconst, 2021-05-07:
```
def recv_callback(line):
  print(line)
  if 'x_min:' in line:
    x_min_open = 'open' in line
    

p.recvcb = recv_callback
p.send_now("M119")
```

lunknowl, 2021-05-07:
> ```
> def recv_callback(line):
>   print(line)
>   if 'x_min:' in line:
>     x_min_open = 'open' in line
>     
> 
> p.recvcb = recv_callback
> p.send_now("M119")
> ```

i know this work but i try to get "x_min_open" outside def

> ```
> def recv_callback(line):
>   print(line)
>   if 'x_min:' in line:
>     x_min_open = 'open' in line
>     
> 
> p.recvcb = recv_callback
> p.send_now("M119")
> print(x_min_open) ---------------- i want use this variable for something else
thanks

volconst, 2021-05-07:
As I noted before, callbacks execute in its own thread later.
Chronologically the assignment of x_min_open is after the place you want to use it in main thread.
You can make the variable readable, but still not initialized with this:
```
def recv_callback(line):
  global x_min_open
  print(line)
  if 'x_min:' in line:
    x_min_open = 'open' in line
```

lunknowl, 2021-05-14:
just a stupid question: i try to read analog value from pin 63 by using "def recv_callback(line):"
PIN:  63   Port: K1 (A 9)  <unused/unknown>   Analog in =   616   Input  = 1  
test = ' Analog in =  ' in line

how do i get the value 616 after analog (value may change) thanks

volconst, 2021-05-14:
```
m = re.search('Analog in = (\d+)', line)

if m:
  print('matched', m.group(1))
```

lunknowl, 2021-05-14:
thanks alot, last question if 1 want to do  p.send_now("M119") then p.send_now("M43 P63"), how can i do it?
def recv_callback(line):
  print(line)

def recv_callback2(line2):
  print(line2)

p.recvcb = recv_callback
p.send_now("M119")

p.recvcb2 = recv_callback2
p.send_now("M43 P63")

volconst, 2021-05-14:
No, p.recvcb2 will not work.

```
recv_ok = False
def recv_callback(line):
  global recv_ok
  if line.startswith('ok'):
    recv_ok = True

def sync_send(cmd):
  global recv_ok
  recv_ok = False
  p.send_now(cmd)
  while not recv_ok:
    time.sleep(0.1)

sync_send("M119")
sync_send("M43 P63")
```

lunknowl, 2021-05-19:
the sync_send doesnt work, it run M119 but not M43p63

closed 2021-05-19 by lunknowl