kliment/Printrun #1193: Pronterface stopped running after an update on my Raspberry Pi 4.

bassamanator, 2021-05-19:
Hi there, 

I installed Pronterface via `sudo apt install printrun`. Everything was running fine (I would often see errors related to python in the terminal though) and I was printing almost constantly with it. I just did a `sudo apt update && sudo apt full-upgrade`, rebooted and now the Pronterface window is no longer showing up. I am able to find it in the task manager however. 

Desperate to get this fixed.

```
Raspberry Pi 4-2GB
Raspberry Pi OS 32 bit
Linux rpi-printer 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
Python 3.7.3
Python 2.7.16
```

volconst, 2021-05-19:
The error log shows python2 is running.
You must run with python 3:
```
python3 /usr/bin/pronterface
```

bassamanator, 2021-05-19:
Thanks! Still not working though.
```
$ python3 /usr/bin/pronterface
   File "/usr/bin/pronterface", line 54
   except getopt.GetoptError, err:
                            ^
   SyntaxError: invalid syntax
```

volconst, 2021-05-19:
Well, you probably have a very old version that uses python2.
Try this https://github.com/kliment/Printrun/issues/879#issuecomment-359346976
I would recommend trying newer releases - latest is preferable but it may require newer dependencies that you do not have.
Try 'pip3 install Printrun' after uninstalling the apt package.

rockstorm101, 2021-05-19:
Hi @bassamanator, what's the output of this command?
```
dpkg -l | grep "pronterface\|printrun"
```

bassamanator, 2021-05-19:
@rockstorm101 
```
dpkg -l | grep "pronterface\|printrun"                                                                                                                                                                               
ii  printrun                              1.6.0-2                                 armhf        3D printing host suite
```

bassamanator, 2021-05-19:
@volconst The installation via pip failed unfortunately https://pastebin.com/7PzWdJ45

Error messages of note (I suspect) are:
>Failed building wheel for wxPython
  Running setup.py clean for wxPython
Failed to build wxPython
...
>Error running configure
    ERROR: failed building wxWidgets
...
>Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-eiply2fc/wxPython/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-dlfe_cdx/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-eiply2fc/wxPython/

I started the install via `python3 -m pip install Printrun`

Let me read that thread you posted, however, I must say that I had switched from stand to tabbed mode in the settings right before this happened.

Appreciate your help.

rockstorm101, 2021-05-19:
> $ dpkg -l | grep [...]

Well that's an old version of Printrun I'm afraid, which is apparently still the version shipped by Raspberry Pi OS. I would recommend [running from source][1]. Sorry about that. Since your version of Raspberry Pi OS seems derived from Debian 10 [these notes][2] will most likely be applicable to your machine too.

[1]: https://github.com/kliment/Printrun#running-from-source
[2]: https://github.com/kliment/Printrun#ubuntudebian-1

kliment, 2021-05-19:
If you switched to tabbed mode, older versions of pronterface (such as 1.6 that you have) will not start. This is a known bug. Delete your .pronsolerc file to get it running again.

bassamanator, 2021-05-19:
@kliment OMG thank you, that did it (deleting .pronsolerc)! Something is up with my printer and I can only print with Pronterface right now! SD card printing and Octoprint result in my print stopping 10 or so minutes after start  (100% failure rate) with an abnormal filament sensor reading error. With Pronterface, however, I've printed several parts this week, with some taking over 8 hours to print. Zero failures. I just don't get it actually. I want to say that there is something wrong with the motherboard and  Pronterface is taking over the 'thinking' somehow but I don't know enough about this technology.

In any case, I'm going to look into what @rockstorm101 posted and see if I can 'run from source', whatever that means. 

Thanks!

bassamanator, 2021-05-19:
So I'm running from source and the version showing in About is 2.0.0rc8 so I guess I'm good to go.

Thanks a bunch guys!

closed 2021-05-19 by bassamanator