kliment/Printrun #1194: ubuntu printcore python using absolute coordinate?

lunknowl, 2021-05-20:
hi, i try to use ubuntu python to send gcode to ramp1.4 marlin using printcore .i use same code on window and window using absolute coordinate but ubuntu using relative?

`from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

p = printcore()
p.connect('COM16', 250000)
while not p.online:
time.sleep(0.1)

p.send_now("G1 X10 Y10 Z0 F24000")
time.sleep(0.7)`

rockstorm101, 2021-05-20:
Maybe try enforcing absolute coordinates as a workaround by sending the 'G90' command? Will look into why the behaviour differs from Windows to Linux. That's interesting.
```
p.send_now("G90")
p.send_now("G1 X10 Y0 Z0 F24000")
```

volconst, 2021-05-25:
Maybe this is related to whether the printer is reset on connection.
You can test whether this happens with the pronterface GUI.
On reset the printer usually sends verbose output of settings - print all received messages to verify reset. 
time.sleep(.7) is guessing, does not guarantee the command is processed - use loop, M400.
Having relative or absolute coordinates should be consistent - probably it is not about relativity, just the printer is not reset and it knows that it is already at X10.
DTR flag toggle of the serial object may help.

closed 2021-06-06 by volconst