kliment/Printrun #1195: Printrun 2.0.0rc5 crashing under MacOS Catalina

ArtBus, 2021-05-26:
I am running the latest version of MacOS Catalina and can't get a single project to complete. Printer gets started and after 5 to 10 minutes the entire computer crashes and reboots. I am using Printrun 2.0.0rc5.

a2k-hanlon, 2021-06-19:
Hi @ArtBus, would you please try the latest release, 2.0.0rc8 as of now, and report back whether of not you have this issue with the newer version? https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8

If the issue is still present for you with 2.0.0rc8, please also provide more details on your setup that may help identify the problem. For example: what printer model you have, what firmware it is running, how it is connected to your computer, the precise version of your OS (for example 10.15.7), etc.

rockstorm101, 2021-09-29:
Hi @ArtBus, any improvements on this issue? Does the latest version work as expected?

a2k-hanlon, 2022-04-19:
Closing due to lack of feedback. @ArtBus if you still have this problem and are willing to provide more information to help resolve it, please feel free to open a new issue.

closed 2022-04-19 by a2k-hanlon