kliment/Printrun #1196: [Question ]return value from inside function

lunknowl, 2021-05-26:
hi i try to return value from inside funtion "def recv_callback(line)" this is my code
def test():
    def recv_callback(line):
        value="test1"
        return(value)
    p.recvcb = recv_callback
    p.send_now("M43 P63")
    return(value)

print(test())

volconst, 2021-05-26:
https://www.datacamp.com/community/tutorials/scope-of-variables-python

lunknowl, 2021-05-26:
this is what i know
```
def outer_func():
    def inner_func():
        test = "hi"
        print(tes)
    inner_func()

outer_func()
```
and this is what i try
```
from printrun.printcore import printcore
from printrun import gcoder
import os
import sys
import time
    
    #-----------------
p = printcore()
p.connect('COM16', 250000)
while not p.online:
    time.sleep(0.1)
        
    #-----------------
def recv_callback(line):
    print("hi")
    test = 1

p.recvcb = recv_callback
p.send_now("M43 P63")
print(test)
time.sleep(0.1)

```

volconst, 2021-05-29:
https://github.com/kliment/Printrun/issues/1192#issuecomment-840689700

closed 2021-06-06 by volconst