kliment/Printrun #1198: GCode Viewer error: gcodeviewpanel has been deleted

tocguy, 2021-06-07:
Just installed Winx64 version 2.0.0rc8 which appears to be a portable version. Anyway, first ever install of Pronterface.

Attempts to view gcode in 3D yields the following:
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted.

Similar error appeared when adding Temperature scale to UI.

Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gui\graph.py", line 391, in StartPlotting
RuntimeError: wrapped C/C++ object of type Graph has been deleted.

volconst, 2021-06-07:
Do these errors break some functionality or are just noise?
My small experience with pronterface on windows is that this occurs only after configuring UI items.
The exceptions are while updating state of components that we no longer needed, so they should be harmless.
If you restart pronterface after configuring UI, I think they will not occur.
I agree that it is not good practice to have such noise, but ... i even can't justify thinking up excuse

tocguy, 2021-06-07:
They prevent the intended function from operating so they are not just noise. I've redownloaded both the 32 bit and 64 bit versions for Windows and the results are the same. I can communicate with the printer but the app itself is of limited utility due to these errors.After writing about the issue I had also tried to load an STL file. It too failed to load but loads ok in Sketchup and Cure so Pronterface just doesn't want to work on my system.

volconst, 2021-06-07:
What exactly does not work? Is it the rendering of the object? Does the object look partial, broken or is entirely missing?
Do these errors occur only after you edit UI settings?
If you restart pronterface after configuring to your preferences, does the rendering work?
Does it matter if you first load the object and then configure or if you configure and then load the object?
Does the STL render if you do not change settings?

tocguy, 2021-06-08:
I understand that you are trying to understand where in the algorithmic chain the process is breaking down but unfortunately, I cannot be of more help. After so many failures I have abandoned hope of using the tool and have deleted it from my system. I know many people utilize the tool so am surprised that it had so many issues attempting to operate on my system.

 

Best of luck,

 

Greg

closed 2021-08-09 by volconst