kliment/Printrun #1199: Micro+ M3d printer connecting

Kivo0, 2021-06-09:
stuck at connecting

volconst, 2021-06-09:
What is the firmware and pronterface version?
Enable Debug Communications from the menu, connect and paste the log here.

Kivo0, 2021-06-14:
Proterface version: Printrun - 2.0.0rc8

Debug enabled console output:

Connecting...
SENT: M105
RECV: e1
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait
RECV: wait

My guess is that the communication of this printer Micro+ is protected by a weird encoding.

also This is the output from the Serial port monitoring software when connecting to The M3D micro+ software:
[14/06/2021 14:13:20] Written data (COM4) 
    4d 31 31 35 0d 0a                                 M115..           
[14/06/2021 14:13:20] Read data (COM4) 
    77 61 69 74 0a 65 31 0a                           wait.e1.         
[14/06/2021 14:13:20] Written data (COM4) 
    82 10 00 00 73 00 06 46                           ‚...s..F         
[14/06/2021 14:13:20] Read data (COM4) 
    6f 6b 20 52 45 50 45 54 49 45 52 5f 50 52 4f 54   ok REPETIER_PROT 
    4f 43 4f 4c 3a 32 20 46 49 52 4d 57 41 52 45 5f   OCOL:2 FIRMWARE_ 
    4e 41 4d 45 3a 4d 69 63 72 6f 2b 20 46 49 52 4d   NAME:Micro+ FIRM 
    57 41 52 45 5f 56 45 52 53 49 4f 4e 3a 32 30 31   WARE_VERSION:201 
    37 30 39 32 30 30 31 20 4d 41 43 48 49 4e 45 5f   7092001 MACHINE_ 
    54 59 50 45 3a 4d 33 44 5f 4d 69 63 72 6f 2b 20   TYPE:M3D_Micro+  
    58 2d 53 45 52 49 41 4c 5f 4e 55 4d 42 45 52 3a   X-SERIAL_NUMBER: 
    53 4b 31 37 30 38 30 39 30 31 30 35 30 30 33 31   SK17080901050031 
    0a                                                .                
[14/06/2021 14:13:20] Written data (COM4) 
    82 10 00 00 1b 00 ad 95                           ‚.....­•         
[14/06/2021 14:13:20] Read data (COM4) 
    4e 6f 74 20 53 44 20 70 72 69 6e 74 69 6e 67 2e   Not SD printing. 
    0a 6f 6b 0a                                       .ok.             
[14/06/2021 14:13:20] Read data (COM4) 
    77 61 69 74 0a                                    wait.            
[14/06/2021 14:13:20] Written data (COM4) 
    84 10 00 00 5b 00 ef 22                           „...[.ï"         
[14/06/2021 14:13:20] Read data (COM4) 
    6f 6b 0a                                          ok.              
[14/06/2021 14:13:20] Written data (COM4) 
    82 16 00 00 6b 02 04 00 00 00 00 0a 88            ‚...k.......ˆ    
[14/06/2021 14:13:20] Read data (COM4) 
    6f 6b 20 50 54 3a 30 20 44 54 3a 2d 31 35 38 37   ok PT:0 DT:-1587 
    34 36 33 35 36 30 0a                              463560.          

I think this is not an issue with your software, this printer is just encrypted. 
So i will close the issue.

closed 2021-06-14 by Kivo0