kliment/Printrun #1200: Reset a 3D printer using pronterface

KiriAuscaf, 2021-06-18:
Hi 

I would like to know how to reset a 3D printer using Pronterface program. Are you able to guide me ?

Thanks 

Kiri

volconst, 2021-06-22:
Isn't there a Reset button? Try disconnect/connect.
Some printers may not restart on connect.
Check DTR setting in pronterface settings.

KiriAuscaf, 2021-06-22:
Hi,

Thank for your response.

The 3D printer Creality 10 3D Max.

Are you able to let me know where is the reset button ?

If you need further information, please don't hesitate to contact me.

Regards
Kiri Loganathan
Product Development Manager

AustralianScaffolds.com.au<https://australianscaffolds.com.au/>
1300 919 905 | 02 9627 1700
***@***.******@***.***>
71 Hobart Street, Riverstone NSW 2765

***@***.***D7680F.4C329320]

Message protected by MailGuard: e-mail anti-virus, anti-spam and content filtering.
http://www.mailguard.com.au/mg

Report this message as spam<https://console.mailguard.com.au/ras/20OKa6O48i/5aCcgCQdLPwNRsI25I7Qzj/2.3>

Message  protected by MailGuard: e-mail anti-virus, anti-spam and content filtering.http://www.mailguard.com.au/mg

KiriAuscaf, 2021-06-22:
Hi ,

Thanks for your response..

The 3D printer Creality 10 3D Max.

Are you able to let me know where is the reset button ?

If you need further information, please don’t hesitate to contact me.

What is the DTR setting meaning ?

Thanks 

Kiri

> Isn't there a Reset button? Try disconnect/connect.
> Some printers may not restart on connect.
> Check DTR setting in pronterface settings.

volconst, 2021-06-24:
Reset button is right after Connect.
DTR is a wire in the serial cable signaling that the printer should restart, but it may not be wired to do so.
DTR is also the 5th option in Settings->Options->Printer Settings. When checked the connect requests printer restart.
Reset button sends DTR separately of connecting.

KiriAuscaf, 2021-06-25:
Hi Team,

Thanks for your response.

When I click the connect button, printer needs to be power on ?

Once it connected  approximately how long it will take to connect ?

Regards
Kiri Loganathan
Product Development Manager

1300 919 905 | 02 9627 1700
***@***.***
71 Hobart Street, Riverstone NSW 2765

***@***.***D769B7.6C89C9A0]

http://www.mailguard.com.au/mg

volconst, 2021-06-27:
> When I click the connect button, printer needs to be power on ? Once it connected approximately how long it will take to connect ?
On some printers, the USB cable is sufficient to power the control board and talk with the printer, but it's mechanical parts won't move. The conventional answer is - yes the printer needs to be powered.
Connection should happen in less than a second. Connecting may take forever if the baud rate is incorrect.

closed 2021-08-09 by volconst