kliment/Printrun #1201: Send gcode to SKRv1.3

lunknowl, 2021-06-21:
hi, i try to send simple gcode to new board skrv1.3
`from printrun.printcore import printcore
from printrun import gcoder
import os
import sys
import time

p = printcore()
p.connect('COM19', 250000)
while not p.online:
    time.sleep(2)

p.send_now("G1 X10 F24000")
time.sleep(0.1)`

but stepper not moving, im able to jog using pronterface software? how can i send gcode using python to this board? this code is working on ramp1.4 thanks

volconst, 2021-06-22:
probably the program exits before the command is sent to/processed by the printer.
Put a longer sleep at the end but better use this code
https://github.com/kliment/Printrun/issues/1192#issuecomment-840689700

lunknowl, 2021-06-23:
i think the skr1.3 is not connect
`from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

#-----------------
p = printcore()
p.connect('COM19', 250000)
while not p.online:
    time.sleep(0.1)

#-----------------
ok_counter = 0
def count_oks(line):
  global ok_counter
  if 'ok' in line:
    ok_counter += 1

def wait_oks(target_count):
  while ok_counter < target_count:
    time.sleep(.1)

p.recvcb = count_oks

ok_counter=0
p.send_now("G1 X10 Y0 Z0 F24000")
wait_oks(1)

ok_counter=0
p.send_now("M114")
wait_oks(1)`

the skr isnt doing anything, waiting for something, but if im using pronterface, it work.

volconst, 2021-06-23:
Some marlin versions does not emit the correct messages on connection.
In pronterface, enable 'Debug communications' in the menu, connect to printer, and paste the log pane contents here.
But if this was the reason, the pronterface GUI and the script would behave the same. Are you using the same codebase for pronterface and the script?

lunknowl, 2021-06-23:
pronterface printrun version 1.6.0, - python printrun version 2.0.0rc7 - the debug information 
`SENT: M105
RECV: ok T:-15.00 /0.00 B:-15.00 /0.00 @:0 B@:0`

volconst, 2021-06-23:
Please include the whole log after enabling "Debug communication" and pressing the connect button.

lunknowl, 2021-06-23:
this is the whole log on skr1.3 board with pronterface printrun v2.0.0rc7
`Connecting...
SENT: M105
RECV: ok T:-15.00 /0.00 B:-15.00 /0.00 @:0 B@:0
Printer is now online.
SENT: M105
RECV: ok T:-15.00 /0.00 B:-15.00 /0.00 @:0 B@:0
`

and this is the script for python on printrun  v2.0.0rc7
`from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

#-----------------
p = printcore()
p.connect('COM19', 250000)
while not p.online:
time.sleep(0.1)

#-----------------
ok_counter = 0
def count_oks(line):
global ok_counter
if 'ok' in line:
ok_counter += 1

def wait_oks(target_count):
while ok_counter < target_count:
time.sleep(.1)

p.recvcb = count_oks

ok_counter=0
p.send_now("G1 X10 Y0 Z0 F24000")
wait_oks(1)

ok_counter=0
p.send_now("M114")
wait_oks(1)`

the log for the script is blank/nothing (waiting for something)

volconst, 2021-06-23:
The script is waiting for printer greeting - usually 'start'.
The log shows that the printer does not greet.
Try pressing pronterface Reset button after you press connect or printer/hardware reset/power button.
Do you have a DTR setting in pronterface 1.6 settings? Is it checked? If not check it and connect again.
Pronterface 1.6 is very old. Install the latest release and enable DTR setting there.

Another approach is to send a M105 command in the script 
```
p.connect('COM19', 250000)
while not p.online:
  p.send_now("M105")
  time.sleep(0.1)
```
This will make the printer send 'ok' and the online flag will be set.
But the problem with this is that ok counting may be wrong - we do not know how many M105 were never delivered to the printer and how many 'ok' we should receive.

Another approach is to remove the while not p.online and leave a greater sleep hoping the connection is established.

lunknowl, 2021-06-23:
well the problem is i can able to connect using pronterface but when using pythons script 
`from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

#-----------------
p = printcore()
p.connect('COM19', 115200)
while not p.online:
    p.send_now("M105")
    time.sleep(0.1)

#-----------------
def recv_callback(line):
    print("line")
        

#-----------------
p.recvcb = recv_callback
p.send_now("M119")`

ERROR:root:Not connected to printer.
ERROR:root:Not connected to printer. 

this is what i get now

volconst, 2021-06-23:
It is expected to get some "Not connected to printer" messages, because we are sending them while we wait the connection.
You can swap the send_now('M105') and sleep, and increase the sleep to reduce the count (to 0).
The script should be able to connect after some error messages.

lunknowl, 2021-06-23:
> It is expected to get some "Not connected to printer" messages, because we are sending them while we wait the connection.
> You can swap the send_now('M105') and sleep, and increase the sleep to reduce the count (to 0).
> The script should be able to connect after some error messages.

i dont think it connect - already 5 mins still ERROR:root:Not connected to printer.

volconst, 2021-06-23:
You should use the same baud rate as pronterface, Your last baud rate is 115200, so it is no surprise that it does not work.

volconst, 2021-06-23:
Move the callback subscription before connect so you see if the printer sends some message
```
def recv_callback(line):
  print("got", line)
p.recvcb = recv_callback
p.connect('COM19', 250000)
```

lunknowl, 2021-06-23:
> You should use the same baud rate as pronterface, Your last baud rate is 115200, so it is no surprise that it does not work.

i using baudrate 115200 as a default for pronterface, python script and firmware, i just saying that on pronterface, i can select different baudrate and it still working

lunknowl, 2021-06-23:
> Move the callback subscription before connect so you see if the printer sends some message
> 
> ```
> def recv_callback(line):
>   print("got", line)
> p.recvcb = recv_callback
> p.connect('COM19', 250000)
> ```

i try and it still say "ERROR:root:Not connected to printer."

volconst, 2021-06-24:
What is your marlin version?
add:
```
p.loud = True
p.connect(...)
```
and paste the log.

Another workaround that may work:
```
p.connect(...)
time.sleep(1)
p.online = True
```

lunknowl, 2021-06-24:
> What is your marlin version?
> add:
> 
> ```
> p.loud = True
> p.connect(...)
> ```
> 
> and paste the log.
> 
> Another workaround that may work:
> 
> ```
> p.connect(...)
> time.sleep(1)
> p.online = True
> ```

hi,im using the marlin 2.0.x,  the "p.online=true" is working, but for Gcode motion only (G1 X10 F100) but not for Mcode (M119), is there different for them?

`from printrun.printcore import printcore
from printrun import gcoder
import sys
import time

#-----------------
p = printcore()

p.connect('COM19', 115200)
time.sleep(0.1)
p.online = True
while not p.online:
    time.sleep(0.1)
    
#-----------------
ok_counter = 0
def count_oks(line):
    global ok_counter
    print(line)
    if 'ok' in line:
        ok_counter += 1

def wait_oks(target_count):
    while ok_counter < target_count:
        time.sleep(0.1)

p.recvcb = count_oks

ok_counter=0
p.send_now("M114")  ///  p.send_now("G1 X10 F100")  is working, thanks
wait_oks(1)
`

volconst, 2021-06-24:
Paste log when you add 
```
p.connect(,,,)
p.loud = True
```

lunknowl, 2021-06-25:
> Paste log when you add
> 
> ```
> p.connect(...)
> p.loud = True
> ```

like i say, blank log, waiting for something, usually "ERROR:root:Not connected to printer"

volconst, 2021-06-25:
> like i say, blank log, waiting for something, usually "ERROR:root:Not connected to printer"
After `p.online = True` you should not see "Not connected to printer"

lunknowl, 2021-06-25:
> > like i say, blank log, waiting for something, usually "ERROR:root:Not connected to printer"
> 
> After `p.online = True` you should not see "Not connected to printer"

no i dont, it just blank like waiting for something, usually when finish script i received a right arror symbol icon ">>> "
but when i run the script it just blank like

Python 
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
=================== RESTART: test.py ===================
>>>  

this is the p.send_now("G1 X10 F100") log

Python 
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
=================== RESTART: test.py ===================

(blank)

this is the p.send_now("M119") log

volconst, 2021-06-25:
Oh, nooo. You are running the code in some interactive interpreter, probably opening the port multiple times in the same process.
Do not do this.
Always run the script in a new process, after the previous process has exited. 
In the cmd window type `python3 myscript.py`
I am not sure if the exiting of the process is sufficient for clearing port state, the problem may arise because of lack of p.disconnect() in the previous process.
The script should include a p.disconnect() either in event handler or at the end of the script
probably in a try ... except KeyboardInterrupt block

lunknowl, 2021-06-25:
"python test.py
_" 
same result
i still think that p.connect not connect or something, but the "G1 X10 F2400" working. really weird.

volconst, 2021-06-25:
Do you see 'ok' for the G1 command.
It is possible that the reading from the printer is broken.
G1 could move the head even if we cannot read from the printer.
The M119 could arrive at the printer as G1, and only the response may have problem.
You also have to be sure that there is no other program that has the port open - can be Cura, pronterface, or the script. Watch you task manager for such processes.

lunknowl, 2021-06-25:
i dont see the ok for G1 command also.

volconst, 2021-06-25:
You have to edit this code
https://github.com/kliment/Printrun/blob/c451359a35d502a8e446702d3cf94cae9e50c0de/printrun/printcore.py#L253-L255

lunknowl, 2021-06-25:
> You have to edit this code
> https://github.com/kliment/Printrun/blob/c451359a35d502a8e446702d3cf94cae9e50c0de/printrun/printcore.py#L253-L255

this is what i change 
```
self.printer = Serial(port = self.port,
                                              baudrate = self.baud,
                                              timeout = None,
                                              parity = PARITY_NONE,
                                              stopbits=1,
                                              bytesize=8,
                                              rtscts=False,
                                              dsrdtr=False,
                                              xonxoff = False
                                              )
```

and i get this error
ERROR:root:Could not connect to COM19 at baudrate 115200:
Serial error: Port is already open.

volconst, 2021-06-25:
If you provide port in the constructor it is automatically opened and after that in the code it is opened again - remove the port parameter.

lunknowl, 2021-06-25:
> If you provide port in the constructor it is automatically opened and after that in the code it is opened again - remove the port parameter.

so i think the port is open but not accept gcode
if i use "p.online = True", Gcode work, Mcode still doesnt work
if i not use "p.online = True", nothing work.

sarmadsiddiqui, 2021-07-22:
I had a similar problem, where I couldn't connect to an SKR v1.3 board. I tried more or less everything mentioned here to no avail. If I connected using the GUI with `pronterface.py`, or with the console using `pronsole.py` I would connect and send commands just fine. After tracing out all the connect commands in the two files, I added the following print statement to 
the beginning of the `connect` function in `printcore.py`:

```
    @locked
    def connect(self, port = None, baud = None, dtr=None):
        """Set port and baudrate if given, then connect to printer
        """
        print(f'sms: port: {port}, baud: {baud}, dtr: {dtr}')
        
        if self.printer:
            self.disconnect()
```

(The pronterface script calls the printcore class for the connection methods). As I never entered a dtr value, it automatically goes through some logic and assigns dtr as `True`. Knowing this, I went back to the example script shows in the **Read Me** but entered the following command (copy/pasting and editing the first post's code here):
```
p = printcore()
p.connect('COM19', 250000, True)
while not p.online:
time.sleep(2)
```

So basically, added the `dtr` arugment as `True` in the connect call. And then all the commands went through as expected. I hope this doesn't cause me problems later, but so far so good.