kliment/Printrun #1204: pid tune problem

xtatorx, 2021-07-26:
replace my hot end with a slice engineering all metal hot end and replaced the  board with the btt skr mine e3 v2 updated marlin firmware and when i try to do the pid tune i keep getting the same error please see attached pic
[image]

kliment, 2021-07-26:
This is probably a hardware problem - I suspect you have a power brownout on your board because your new hotend pulls too much power, and it's causing the microcontroller to reset itself and the USB port to no longer work. Sadly there's nothing we can do about it in software.

xtatorx, 2021-07-26:
how do i know forsure and whats the fix then

xtatorx, 2021-07-26:
so i was able to get controls to work,  pronterface and printer communicating but when i start the pid tune this happens.  

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range

kliment, 2021-07-27:
What did you change to make it work?

That last error is something on our side. Can you enable "settings" -> "debug communications" and run it again?

Also, which version of printrun are you running?

volconst, 2021-07-28:
@xtatorx you can test whether artifacts at https://github.com/volconst/Printrun/actions/runs/1074748808 fix the problem. The are 2 exceptions and the fix is only for the last, but the other may be ok to occur.

xtatorx, 2021-07-29:
running debug now and its the 2.0.0rc8

xtatorx, 2021-07-29:
THIS IS WHAT HAPPENED AFTER THE DEBUG RUN
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range

xtatorx, 2021-09-14:
how do i fix this

xtatorx, 2021-09-14:
> @xtatorx you can test whether the artifacts at https://github.com/volconst/Printrun/actions/runs/1074748808 fix the problem. There are 2 exceptions and the fix is only for the last, but the other may be ok to occur.

tried this with no fix

xtatorx, 2021-09-14:
Connecting...
Printer is now online.
>>> m303 e0 s200 u1
SENDING:M303 E0 S200 U1
PID Autotune start
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range
Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "printrun\printcore.py", line 744, in _send
  File "serial\serialwin32.py", line 317, in write
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

DivingDuck, 2021-09-14:
Hi @volconst, 
I just recognize the parameter **u**. Do printrun decode this parameter for PID tuning?
M303 E0 S200 **U1**

xtatorx, 2021-09-14:
tried that. with no luck

DivingDuck, 2021-09-14:
what happen if you do:
 M303 E0 S200
without U parameter?

xtatorx, 2021-09-14:
same error

volconst, 2021-09-15:
@xtatorx , Enable debug communication in the exe from https://github.com/volconst/Printrun/actions/runs/1074748808
and include the log together with the exception stack trace.

volconst, 2021-09-15:
> Hi @volconst,
> I just recognize the parameter **u**. Do printrun decode this parameter for PID tuning?
> M303 E0 S200 **U1**

I believe unknown parameters are forwarded transparently to the printer.

xtatorx, 2021-09-15:
i did the debug file abd posted it to my post on github

volconst, 2021-09-15:
@xtatorx ,
> i did the debug file abd posted it to my post on github

> File "printrun\printcore.py", line 354, in _readline
> IndexError: tuple index out of range

The stack trace line number "354, in _readline" corresponds to rc8
https://github.com/kliment/Printrun/blob/5d0ec6d25dd67fc51afccf028d4e77606ec81900/printrun/printcore.py#L354

I instructed you to (download, run and) test exe from https://github.com/volconst/Printrun/actions/runs/1074748808
whose line 354 does not seem capable of throwing the index error you report

https://github.com/volconst/Printrun/blob/1b1d1b531c74373d5b69bce73b204ebc8c5e2f93/printrun/printcore.py#L354

Please verify that you downloaded and ran the fixed exe and include it's log.

xtatorx, 2021-09-15:
i downloaded that one and ran it but ill get the debug logs

xtatorx, 2021-09-18:
trying to get the debug logs now  
with that exe you said to use

xtatorx, 2021-09-18:
> @xtatorx ,
>
>
>
> The stack trace line number "354, in _readline" corresponds to rc8
> https://github.com/kliment/Printrun/blob/5d0ec6d25dd67fc51afccf028d4e77606ec81900/printrun/printcore.py#L354
>
> I [...]

how do i get the log   thats the version im using fromt he exe link you sent

xtatorx, 2021-09-18:
[image]

xtatorx, 2021-09-18:
im also getting a heating failed  E1 printer halted.  what does that mean.  im using the stock ender 3 pro heater cartridge and a e3d thermistor.

volconst, 2021-09-20:
@xtatorx , thanks for the feedback.
I now see that the second (index) exception is not raised and that's a progress for pronterface.
The E1 in the message you get probably means Extruder1 (that is a second extruder), which is strange because you typed e0 in the command.
You have to check your marlin configuration - how many extruders are configured.
I you do not have it, in the marlin source code are available many config files, probably there is one for your model, you have to customize it, see the extruder configuration there, compile and  flash the binary to the printer board.
This is complex, and before this you can check the wirings, did you connect to the correct pins.
You can just check if heating works - set a target temperature to 40 degrees C, and touch the heater block to feel if it heats. This will tell you if heating wires are ok, and also the thermistor to some extent. Beware that if something is wrong it may get dangerously hot. Restart the printer after this, to prevent further heating.
Another workaround is to put the m303 command in a gcode file, copy it to a SD card and print it from the sdcard, with the usb cable disconnected to prevent any doubt of pronterface involvement.

xtatorx, 2021-09-20:
the hot end is heating up. i do know that. i can heat the printer to 240c
and it stays there. when i go above that is when i get the e1 error. i have
a feeling its because i cant do the pid tune.  so how do i fix the pid
issue first

volconst, 2021-09-21:
@xtatorx 
I think the printer halts because it does not reach the desired temperature in some hard coded time in marlin.
You can probably reach higher temperatures if you put silicon sleeve on the heater block.
However I think that the issue is not pronterface related, especially if the printer lcd says - printer halted.
If you still think that the problem is in pronterface, please include screenshot of the printer lcd with your post.

xtatorx, 2021-09-21:
i know the heating failed isnt in pronterface. but when i try to do a pid
tune in pronterface thinking that may help with the heating failed error,
it diaconnects and throws some codes i dont understand. see the pics.  i
think the heating failes may be a bad heater cartidge. becuase it wont go
above 240.   ive already replaced the thermistor and changed marlin for
it.   so i have 2 problems. really.

1. cant do pid tune
2.  heating failed.

wasnt sure if one was causing the other.  i know very little about software
and coding.

volconst, 2021-09-21:
Is the printer halted when the PID autotune fails?
Please post photo of the printer lcd when the PID autotune fails.
You can also enable the pronterface temperature graph in settings and post a screenshot of the temp graph and error pane with the exception together.
What is the marlin version?
I told you before to put the m303 command in sd/tf card and print the file from it without usb cable connected.
Please reply with the effect of all my tips, or I will lose interest and leave the conversation.
It is possible that marlin have maximum temp and refuse going above it. It should be checked whats the max temp value, it is probably a marlin directive.
Are you able to compile and flash marlin?

volconst, 2021-09-21:
> i know the heating failed isnt in pronterface. but when i try to do a pid tune in pronterface thinking that may help with the heating failed error,
If the printer can't reach 260 deg C, e.g. due to power, then it is logical that the PID tune also fails, because the tune just measures how fast the heater reaches the temp and cools down.

xtatorx, 2021-09-21:
> Is the printer halted when the PID autotune fails?
> Please post photo of the printer lcd when the PID autotune fails.
> You can also enable the pronterface temperature graph in settings and post a [...]

no the printer is not halted when pid tune fails.  im not sure how to put the m303 on sd. can you please explain.. i have already changed all the marlin max temp settings to 285c.   im using the marlin 2.0.08 firmware. and see attachted pics for graph lcd and error message

xtatorx, 2021-09-21:
[image]
[image]
[image]

xtatorx, 2021-09-21:
> If the printer can't reach 260 deg C, e.g. due to power, then it is logical that the PID tune also fails, because the tune just measures how fast the heater reaches the temp and cools down.

i will order a new heater cartridge.  how do i save the m303 to sd card not sure of that

volconst, 2021-09-22:
> how do i save the m303 to sd card not sure of that

Open notepad.exe
paste your command, e.g.:
```
m303 e0 s200
```
save file as pidtune.gcode
Verify that the file is not saved as .txt file. If saved with .txt extension, remove it, so the filename ends with ".gcode"
copy to sd card.
eject sd card (see if there is eject command in the context menu of the sd card drive letter)
move sd card to printer, disconnect printer, restart, print the file. Watch the lcd for signs that the printer halts or restarts and tell me the result.

I now see there is debug option in m303 command https://marlinfw.org/docs/gcode/M303.html
It may be useful when sending m303 from pronterface.
```
m303 d1 e0 s200
```
But it may be disabled, then you have to uncomment PID_DEBUG in Configuration.h, compile and flash.
Send the command and watch if it generates debug info in the log pane.

xtatorx, 2021-09-22:
thanks i work on that and see what happens. give me a day