kliment/Printrun #1207: On 2.0.0 UI completly broken on RTL locals

thebestnom, 2021-08-01:
On 2.0.0 UI completly broken on RTL locals

changing the local to english in the code works

example
[image]

volconst, 2021-08-04:
Looks good to me. Yes, it is unusable, but isn't this the idea of RTL by design? :grin: 
How do you switch to RTL?

BorisKozo, 2022-06-18:
@thebestnom Can you elaborate what you changed? I am having the same issue.

BorisKozo, 2022-06-19:
OK it is more broken than I initially thought. I tried to put a locale En-il and the application won't even start.
With the RTL UI some of the buttons are broken and it makes the application unusable. Currently I set my computer locale to En-Uk just to be able to use it.

DivingDuck, 2022-06-19:
Good morning,
purely out of interest what OS and language do you use?

BorisKozo, 2022-06-19:
Windows 10, everything is English except the locale which is Hebrew (Israel).
If I set the locale to English (Israel) it will crash on start.
Now I set it to English (United Kingdom) and everything works normal.

DivingDuck, 2022-06-19:
For the crash issue I maybe can help you. This seems to be a known problem with wxPython. In case you are able the compile pronterface at your own you can use the latest developer snapshot version of wxPython. Replace the 4.1.x version via pip install:

`pip install -U --pre -f https://wxpython.org/Phoenix/snapshot-builds/ wxPython --use-deprecated=html5lib`

Alternatively you can test the interim version from my test branch: 
https://github.com/DivingDuck/Printrun/actions/runs/2525453711

BorisKozo, 2022-06-20:
@DivingDuck Thank you for your help. I am currently OK with keeping my computer at UK locale but I hope the whole locale detection will get fixed. We don't expect the UI to work well in RTL but also don't go to that mode at all.

thebestnom, 2022-06-20:
Btw, there is a way to fix with changing the code without changing the whole system