kliment/Printrun #1208: custom button doesn't work

TZYaaa, 2021-08-05:
When I click the custom button such as 'extrude 5mm', the extruder doesn't move or even retracts. The other extruding custom buttons and retracting custom buttons also work disorderly.

volconst, 2021-08-06:
Enable Settings/Debug comminications and try the buttons. The output in the log should contain the commands.
What version is pronterface, OS?
Include the code of the custom buttons.

TZYaaa, 2021-08-08:
Thank you for commenting!
It's Windows version. The code is 'G1 E5 F3000‘.

volconst, 2021-08-09:
Please include the output after you Enable Settings/Debug comminications and click the button.
You have to ensure that the COM port is opened/used only by pronterface - you may need to close all other 3d printing slicers, hosts (e.g. Cura) and even check for another (windowless) pronterface process in windows task manager.

TZYaaa, 2021-08-10:
The output is 'SENT: M105 RECV: ok T:20.3 B:22.0'
I checked and operated as your suggestion, but that still doesn't work.

DivingDuck, 2021-08-10:
You are right. This happen if you forget to set your extruder in abs / relative mode before extruding. I can replicate this with my smoothie board. In case you put G91 in front of your code you can extrude a length repeatably with you custom button.
In case you choose abs (G90) you can do this only once as the position is the same (this is what you experience if you start your printer and do the command first time)

Try  'G91 G1 E5 F3000‘.

TZYaaa, 2021-08-11:
Thank you for commenting! I tried the 'G91 G1 E5 F3000‘, but that still doesn't work.

DivingDuck, 2021-08-11:
The only other possibility I can think about for this is thermal runaway protection. Try to heat up your heater to 180 °C before doing the command.

TZYaaa, 2021-08-11:
My printer is r3bEL bioprinter. Two screen shots are attached. Thank you!
[image]
[image]

DivingDuck, 2021-08-11:
Nice. I have seen those some when in the 2018. They are using reprap firmware if I remember correct.

Anyway, guess, you are maybe on Windows10. You should update to the actual RC version of pronterface as the old versions won't work very well with actual windows 10 releases. Before doing this you should backup your original printrunconf.ini. It is located in your user directory.

And forget my comment on thermal runaway as I guess this isn't really an issue for this kind of printer :)

volconst, 2021-08-11:
@TZYaaa 
> Thank you for commenting! I tried the 'G91 G1 E5 F3000‘, but that still doesn't work.

```G91 G1 E5 F3000```
on a sigle line does not look like correct syntax.
It should be 2 separate lines, as it is 2 commands
```
G91
G1 E5 F3000
```
As our attention is already on relativity and your custom button does something already implemented you can Enable again Debug Communications and try the stock Extrude button.
You should see the commands it uses and copy them to your custom button.

TZYaaa, 2021-08-12:
I try to install the latest version Printrun-2.0.0rc8, but an error message occurs showing 'Failed to execute script pronterface'.

DivingDuck, 2021-08-12:
The wx problem, I guss.

Can you try to check this version?
**for Windows10 x64**: https://github.com/DivingDuck/Printrun/suites/3478243774/artifacts/82728991
**for Windows10 x32**: https://github.com/DivingDuck/Printrun/suites/3478243774/artifacts/82728994
The Differences are:

- include the latest developer version of  wx-python, which should eliminate the execution problem
- include the GTK for projector module

TZYaaa, 2021-08-12:
Thank you! This version is installed successfully. However, the custom button still doesn't work.

DivingDuck, 2021-08-12:
Sorry, I accidental close your issue.

Can you attach the  printrunconf.ini and the log file? You need to enable creation the log file via settings -> options
[image]

volconst, 2021-08-12:
> Thank you! This version is installed successfully. However, the custom button still doesn't work.

Did you made the custom button use the 2 lines I provided?
Test to send the code of the custom button manually with the Send button.

TZYaaa, 2021-08-12:
The below is the printrunconf.ini:
set last_window_maximized True
set last_window_width 805
set last_window_height 529
set last_sash_position 586
set port COM3
button 0 "Set Home" /c "#cccccc" G92 X25 Y25 Z0
button 1 "Set Z0" /c "#cccccc" G92 Z0
button 2 "Get position" /c "#cccccc" M114
button 3 "Move to the Front" /c "#cccccc" G1 X100 Y10 F6000
button 4 "P0" /c "#66b3ff" G1 X25 Y25 F6000
button 5 "P1" /c "#66b3ff" G1 X25 Y175 F6000
button 6 "P2" /c "#66b3ff" G1 X175 Y175 F6000
button 7 "P3" /c "#66b3ff" G1 X175 Y25 F6000
button 8 "Extrude 150 mm" /c "#33cc33" G1 E150 F3000
button 9 "Extrude 50 mm" /c "#33cc33" G1 E50 F3000
button 10 "Extrude 5 mm" /c "#33cc33" G1 E5 F3000
button 11 "Extrude 1 mm" /c "#33cc33" G1 E1 F3000
button 12 "Retract 150 mm" /c "#ff4d4d" G1 E-150 F3000
button 13 "Retract 50 mm" /c "#ff4d4d" G1 E-50 F3000
button 14 "Retract 5mm" /c "#ff4d4d" G1 E-5 F3000
button 15 "Retract 1mm" /c "#ff4d4d" G1 E-1 F3000
set last_file_filter 0
set last_file_path C:\Users\Tong Research Group\Desktop\PC backup\Bioprinter files\Trials\Random

TZYaaa, 2021-08-12:
[image]
There is no log file, I guess.

TZYaaa, 2021-08-12:
@volconst 
Thank you, volconst! I also tried, but that doesn't work.

DivingDuck, 2021-08-12:
You need to put in a path and file name like:
C:\Users\Tong Research Group\printrun.log
Then restart pronterface and it will create a log file with all activities in the desired location.
I add an button to my profile and run the commands. Works well with my printer (firmware is: smoothieware)

[image]

I'm miss some information like baud rate, build dimensions, feed rates in your ini file. Don't know if this matter.

Below I attach my printrunconf.ini and resulting log file for comparison. You can use my ini file (quit pronterface program, make a backup of your file and then copy it to your location and restart pronterface) for testing the button, but don't print with it as my bed is huge and my printer setup will not match your printer. Maybe it will help you.

[printrunconf.zip](https://github.com/kliment/Printrun/files/6975460/printrunconf.zip)

volconst, 2021-08-12:
> @volconst
> Thank you, volconst! I also tried, but that doesn't work.

It appears that buttons can have only one line.
Define a macro containing the 2 lines and then a button having the macro name as a command.

DivingDuck, 2021-08-12:
@volconst is this a firmware thing? For smoothieware it works fine.

volconst, 2021-08-12:
@DivingDuck 
the pronterface UI for defining button command is a single line textbox.
I am not sure what happens if you add several commands with separators e.g. ':' or ';' .
It surprises me that your command line 'G91 G1 E5 F3000‘ works for you.
I have never used multiple commands per line.

volconst, 2021-08-12:
@TZYaaa  may Try 'G91 : G1 E5 F3000‘ or 'G91 ; G1 E5 F3000‘ for the button command without defining macro.

DivingDuck, 2021-08-13:
@volconst
I found this some years ago. I use this for changing filament. It is maybe worth to check this for other firmware's too. 
One can easily check this for own printers using first the absolute mode 'G90 G1 E5 F3000'. This should run the motor only once if you do the command multiple times. Then switch back to 'G91 G1 E5 F3000' and the motor should each time again if you do the command multiple times.

Adding a semicolon ';' is maybe not a good idea as this is the identifier for comments and gcode interpreter commonly will ignore everything behind a semicolon for a gcode line of code. Colon and dot works at least for smoothieware in this example too. I think they parse the line and ignore everything what is not matching a gcode command. I can also do things like eg.  'G91 G1 E5 F3000 G90 G1 E5 F3000' in one line.

TZYaaa, 2021-08-13:
@volconst 
Sorry, volconst. That still doesn't work.

TZYaaa, 2021-08-13:
@DivingDuck 
Ok, I will try this weekend. Thank you for so detailed clarification!

DivingDuck, 2021-08-13:
@TZYaaa
For testing it is maybe a good idea using a bigger value than 5 mm as this isn't a lot with F3000 in combination. This takes for my motor more or less a second. Quite short for recognizing a movement.
Good luck!

DivingDuck, 2021-08-16:
Don't worry, asking is always better.
Yes, you can do it for testing the last button I have created for this purpose. Save your own one before you copy my version. You will see that Pronterface Gui will look a bit different (as you can see in my attachment above 4 days ago).

TZYaaa, 2021-08-16:
@DivingDuck 
I did as your suggestion, and it really changes (attached), but it isn't your version.
[image]

DivingDuck, 2021-08-20:
Hello @TZYaaa,
Sorry for my late answer. I was traveling and busy the last days and hadn't recognized that you edited your answer.
I'm a bit confused about what version of Pronterface you are playing with. Your attached picture shows that you don't use my config file and the error indicate that there is something wrong with your button definition (maybe with your color definition).

Please attach your config file (you need first archive the file with a tool like 7zip, Zip,  Winrar and then attach this archive file here).

DivingDuck, 2021-09-02:
@TZYaaa
Had you solve your problem?

TZYaaa, 2021-09-02:
@DivingDuck 
I am so sorry that I am unavailable for one month. I will be try that and reply to you next week. Thank you for helping!

DivingDuck, 2021-09-02:
No problem, you are welcome.

TZYaaa, 2021-09-06:
[printrunconf.zip](https://github.com/kliment/Printrun/files/7113328/printrunconf.zip)
@DivingDuck 
 I just delete my original config file and use yours, but I don't know why the pronterface shows like that. I have also checked my computer, it shows there is only your config file.

DivingDuck, 2021-09-06:
For windows there are two places where you can find the config file. Replace YOUR_USER_NAME with your user name.

Place one:
**c:\Users\YOUR_USER_NAME\printrunconf.ini**

Place two (usually a hidden directory):
**c:\Users\YOUR_USER_NAME\AppData\Local\Printrun\Printrun\printrunconf.ini**

I just recognize, that all user path's in printrunconf.ini are set to my user name ...\Armin\...
Please change this to your user name, **especially for the log file path "set log_path c:\Users\Armin\Printrun.log" at line 34**, as this wrong entry will cause  for your computer / user name  an unhandled exception like shown here:

[image]

Guess, this is the main problem between my printrunconf.ini file and your setup. Sorry, I forgot to mention this.

TZYaaa, 2021-09-08:
@DivingDuck 
Thank you for your instruction. I find
[image]
 my config. file. But the username of your config. file shows its mine.

DivingDuck, 2021-09-08:
Here is a modified version of my printrunconf.ini for your test, I changed the path to your user location. Guess this will make it a little bit easier

Please  proceed the following actions: 

1. Close Pronterface in case it is running.
2. Unzip printrun.ini from attached zip file.
3. Copy this file into Directory:
    **c:\Users\Tong Research Group\printrunconf.ini**
3. Delete directory:
    **c:\Users\YOUR_USER_NAME\AppData\Local\Printrun**
4. Start Pronterface

The program should look now similar to my screenshot above and you can see the custom button I made for you. Connect the printer via connect button and then test the button "TZYaaa extrude 5mm". Extrude should hopefully will work now. 

Please post a screenshot and attach the log as zip file so that I can see what happen. Location for it is:
   **c:\Users\Tong Research Group\Printrun.log**

If everything went well you then need to create a new setup file for your device. The steps are the following:
1. Close the program in case it is running
2. Delete my config file **c:\Users\Tong Research Group\printrunconf.ini**
3. Delete directory **c:\Users\Tong Research Group\AppData\Local\Printrun**
4. Start Pronterface
5. Open menu ->Setting ->Options ->Printer Settings and setup your device.
    Com port, Baud rate, build dimensions and Extruder counts should match with your device.
    Check Monitor printer status as this is helpful in the beginning
6. Switch to ->User interface and add Log path: **c:\Users\Tong Research Group\Printrun.log**
    For Interface- and Controls mode choose Standard, all other is up to you.
7. Test connection, home your x, y and z axis with the GUI command buttons and do some slight test movements with all axis.
    Set Length and speed to an appropriate value (e.g.  5 or 10 mm and speed to 80 mm/min), 
    hit the Extrude button and see what happen. Reverse button will reverse the extrusion.
    All steps up to here should work before you go further.
8. Now you are ready to define your custom buttons
    Hit the +button and create your first button. Like:
       Button title: **Extrude 5mm**
       Command: **G91 G1 E5 F80**
       Color: **#DCE480**
     Hit Ok button
     Remark: For testing it is maybe better to lower the speed to something like 80mm/min or even lower so that you are able to hear the motor spinning. You can change it later back to a value you need for normal conditions.

     Same command as Macro button looks like:
     1. Create a macro via menu: ->Settings ->Macro ->new
         Macro Name: **Extrude5mm** (without blanks)
         Add two lines:         
         **G91**
         **G1 E5 F80**
         Hit Save button
      2. Hit the +button for a new custom button 
            Button title: **Extrude 5mm**
            Command: **Extrude5mm**
            Color: **#DCE480**
          Hit Ok button

[printrunconf.zip](https://github.com/kliment/Printrun/files/7130926/printrunconf.zip)

TZYaaa, 2021-09-13:
@DivingDuck 
Thank you so much for so detailed instruction! I am busy these days, I will try a little bit later. Thank you!

TZYaaa, 2021-11-02:
@DivingDuck 
Dear DivingDuck, so many thanks for your detailed instruction. The pronterface can work well now! Thank you again for that!

DivingDuck, 2021-11-03:
@TZYaaa 
It's a pleasure to hear you have manage to solve the problems and pronterface is working now as you expect it. I wish you interesting times with your bio printer.

DivingDuck, 2021-11-03:
Problem solved

closed 2021-11-03 by DivingDuck