kliment/Printrun #1211: new version M43 "Got rubbish reply from" issue

lunknowl, 2021-08-18:
hi, i have question with pronterface and trigorilla board 1.4 
1) if im using pronterface 2.0.0rc8 i can connect to the board no problem but when im send "M43 P3" i get 
```
Got rubbish reply from COM7 at baudrate 250000:
Maybe a bad baudrate?

```
but if i send  "M43 P43" it work fine
2) if im using pronterface 1.6.0, sometime "connecting" forever, i need to use "2.0.0rc8"to connect first then use "1.6.0" for "M43 P3"
any different from these version that either version"2.0.0rc8" can use "M43 P3" or version "1.6.0" connection setting like version "2.0.0rc8"

volconst, 2021-08-18:
https://github.com/kliment/Printrun/blob/82ed6302507eff93247e931c77f1c16b1a71f7fd/printrun/printcore.py#L349-L351
This is caused by UnicodeDecodeError.
Put a print(line_bytes) before https://github.com/kliment/Printrun/blob/82ed6302507eff93247e931c77f1c16b1a71f7fd/printrun/printcore.py#L337
to see the bytes. It could be some compression algo, or non utf-8 text.

voltrare, 2024-09-11:
Hi, im having the same issue, Im using the same board

DivingDuck, 2024-09-12:
Hi @voltrare,

Please update to the actual version. We are in 2024 and this thread is from 2021 (some versions behind the actual version). In addition we need to know a little bit more for an error report than "me too". This won't tell us in the end anything specific. 

For an error report we need to know what operating system, printer hardware, printer firmware and Pronterface version you are running and of cause at what baud rate you are using for communication. This should be the same as in your firmware setup. In addition a error log would be helpful too. 
Maybe it is better to open a new report, as a release candidate will not be updated and the actual version of Pronterface is 2.1
Best regards,
DD

RebelSoftware, 2025-02-07:
Same issue using pronterface 2.2.0 on windows 10 talking to anycubic mega i3 (trigorrilla 1)

Cura talks to the printer just fine. Cura is not running and computer rebooted since running cura.

I get the message and then a continuous list of "sent: M105" in the debug output
If I try to send gcode to the printer I get 
"printer is not online"

I have tried changing the baudrate so that the virtual usb driver and pronterface are using the same rate

RebelSoftware, 2025-02-07:
My problem has been resolved. I had assumed a lower baudrate. once I took it up to 250K it responded straight away. Sorry for the reported error. Clearly a problem with the carbon based unit.

closed 2025-02-07 by DivingDuck