kliment/Printrun #1213: cant pid tune

xtatorx, 2021-09-14:
i have an ender 3 pro with a skr mini e3 v2 board with marlin firmware, slice engineering all metal hot end, bondtech direct drive and i cant pid tune it. I keep getting the same error everytime.  please help

Connecting...
Printer is now online.
>>> m303 e0 s200 u1
SENDING:M303 E0 S200 U1
PID Autotune start
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range
Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "printrun\printcore.py", line 744, in _send
  File "serial\serialwin32.py", line 317, in write
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

volconst, 2021-09-14:
Duplicate of #1204
You did not answer to https://github.com/kliment/Printrun/issues/1204#issuecomment-888450838

xtatorx, 2021-09-14:
?

closed 2021-09-14 by xtatorx