kliment/Printrun #1214: pid tune error

xtatorx, 2021-09-14:
not sure what is wrong when im trying to do the pid tune on my ender 3 pro with slice engineer all metal hotend and skr mini e3 v2 board with marlin software.   what do i do.

Connecting...
Printer is now online.
>>> m303 e0 s200 u1
SENDING:M303 E0 S200 U1
PID Autotune start
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range
Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "printrun\printcore.py", line 744, in _send
  File "serial\serialwin32.py", line 317, in write
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

DivingDuck, 2021-09-14:
Duplicate of #1213, #1204

Please do not open multiple times an issue report to identical issues. Thanks

closed 2021-09-14 by DivingDuck

xtatorx, 2021-09-14:
i deleted one