kliment/Printrun #1216: Keyboard jogging size issue

d-kozminihs, 2021-09-24:
Hi. Just discovered this great interface. Having a problem - keyboard  jogging (KJ) is different from circle jogging. When I activate KJ to 1mm steps it actually does much smaller travel; at 10mm it does around 2mm, at 100mm it travels around 20mm. On-screen jogging works perfectly fine 10=10, 100=100, but KJ is different. Thanks

volconst, 2021-09-29:
What version and OS are you using? Try using the latest 
The normal steps the buttons make are 0.1, 1, 10, 100.
If you hold shift while pressing arrows then the steps are doubled. With control key, they are halved.
It seems that you are pressing shift and at the same time the button units are wrong.
Enable Settings->Debug communications and observe the commands that are actually sent. Are the G0 X numbers correct?
Carefully track which arrows, steps you activate and paste the log here.

d-kozminihs, 2021-09-30:
Ok.. I have tested few things now. Shift and CTRL keys have no effect on that computer (Win7). Debugging indicates (and this is proven with measurements) that all the outputs are exactly 10x smaller than intended -  0.01, 0.1, 1, 10

**Keyboard key pressed three times at 100mm scale position (real travel measuread as 10mm):**
SENT: G91
SENT: G0 Y-10.0 F1000
SENT: G90
RECV: ok
SENT: M114
RECV: ok
RECV: ok
RECV: X:100.00 Y:70.00 Z:0.00 E:0.00 Count X:2655 Y:1859 Z:0
RECV: ok
SENT: G91
SENT: G0 Y-10.0 F1000
RECV: ok
SENT: G90
SENT: M114
RECV: ok
RECV: ok
RECV: X:100.00 Y:60.00 Z:0.00 E:0.00 Count X:2655 Y:1593 Z:0
RECV: ok
SENT: G91
SENT: G0 Y-10.0 F1000
RECV: ok
SENT: G90
SENT: M114
RECV: ok
RECV: ok
RECV: X:100.00 Y:50.00 Z:0.00 E:0.00 Count X:2655 Y:1328 Z:0
RECV: ok

Not quite sure how to interpret the situation with inactive Shift and CTRL keys... Will try a different keyboard

volconst, 2021-09-30:
What is your pronterface version?

d-kozminihs, 2021-10-01:
I have installed this version: 
pronterface-windows-x64_3.8-2.0.0rc8.zip

volconst, 2021-10-02:
What is your screen resolution?
What's the output from this command:
```
!print(self.GetContentScaleFactor())
```

d-kozminihs, 2021-10-06:
Screen resolution is 1280x1024.
Output as follows:

>>> !print(self.GetContentScaleFactor())
Traceback (most recent call last):
  File "printrun\pronterface.pyc", line 703, in sendline
  File "cmd.pyc", line 221, in onecmd
  File "printrun\pronsole.pyc", line 352, in do_shell
  File "<string>", line 1, in <module>
AttributeError: 'PronterWindow' object has no attribute 'GetContentScaleFactor'

Doesn't seem to recognise that command correctly.