kliment/Printrun #1217: ini file printface

ken58w, 2021-09-28:
its not making the ini file for wifi why what is going on?

volconst, 2021-09-29:
This message does not come from Pronterface, just received by it.
It comes either from your printer or your router - you should check their INI.

ken58w, 2021-09-29:
its a wifi card sd unit  not a ptinter

ken58w, 2021-09-29:
it is not printer or router it is a FYSETC wifi sd card unit SD-WIFI with Card-Reader Module run ESP web Dev Onboard the utube demo show it programing card https://www.youtube.com/watch?v=nHNZPRl8gzA

volconst, 2021-09-29:
See https://github.com/FYSETC/FYSETC-SD-WIFI
Fill the wifi network name and pass in this file https://github.com/FYSETC/ESPWebDAV/blob/fysetc/ini/SETUP.INI
and copy it to the card root (either by ordinary card reader or the non USB2UART switch state, or use M50,M51,M52 commands to connect, then map network drive)