kliment/Printrun #1222: Nothing to connect to when plugged into printer.

Tundesser, 2021-11-04:
When I connect my USB cable from printer to laptop, Ponterface only shows a blank connection for communications. It shows no other option. When I connected for the first time, My Windows 10 Home laptop recognized a USB connection and stated it arranged for new device, yet it shows nothing in Pronterface. I've disconnected, powered down, and repositioned USB cable with no success yet. Does anyone recognize this and help? Thank You.

volconst, 2021-11-04:
Thanks for updates.

kliment, 2021-11-04:
Just to clarify in case someone else has this issue - on Windows and Mac OS there are many USB to serial converters that don't have drivers out of the box. This is the most common reason for USB ports not showing on those operating systems. Your printer or printer controller board manufacturer will almost always have drivers linked from the instructions, so that's the first place to check.

closed 2021-11-04 by kliment