kliment/Printrun #1225: Pronterface crashes under python 3.10

kapanala, 2021-11-14:
Running supplied pronterface on Fedora 35:

$ pronterface
WARNING:root:RPC server bound on non-default port 7979
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 62, in <module>
    app = PronterApp(False)
  File "/usr/lib64/python3.10/site-packages/printrun/pronterface.py", line 2388, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib64/python3.10/site-packages/printrun/pronterface.py", line 218, in __init__
    self.reload_ui()
  File "/usr/lib64/python3.10/site-packages/printrun/pronterface.py", line 294, in reload_ui
    self.createGui(self.settings.uimode == _("Compact"),
  File "/usr/lib64/python3.10/site-packages/printrun/gui/__init__.py", line 255, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "/usr/lib64/python3.10/site-packages/printrun/gui/viz.py", line 103, in __init__
    root.gwindow = gviz.GvizWindow(build_dimensions = root.build_dimensions_list,
  File "/usr/lib64/python3.10/site-packages/printrun/gviz.py", line 77, in __init__
    self.p = Gviz(panel, size = size, build_dimensions = build_dimensions, grid = grid, extrusion_width = extrusion_width, bgcolor = bgcolor, realparent = self)
  File "/usr/lib64/python3.10/site-packages/printrun/gviz.py", line 196, in __init__
    self.mainpen = wx.Pen(wx.Colour(0, 0, 0), penwidth)
TypeError: Pen(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 2 has unexpected type 'float'
  overload 3: argument 1 has unexpected type 'Colour'

Generating a local environment using python 3.9, and installing wx and printrun using pip, pronterface sort of runs with a few GTK errors.   I have to assume the issue is the version of python.   However, I don't know for certain whether it is caused by different versions of printrun.  I have not managed yet to install 2.0.0rc7 under python 3.10.  Fedora distribution provides 2.0.0rc5.

hroncok, 2021-11-15:
I've started fixing this in https://github.com/kliment/Printrun/pull/1224 even before seeing this report. Are you able to test that PR?

hroncok, 2021-11-15:
I will likely update the package to the latest rc and include the patch at the same time.

hroncok, 2021-11-16:
@kapanala Could you please test this update?

```
sudo dnf --setopt=install_weak_deps=False install https://kojipkgs.fedoraproject.org//work/tasks/6157/78956157/printrun-2.0.0-0.20.rc8.fc35.x86_64.rpm https://kojipkgs.fedoraproject.org//work/tasks/6157/78956157/printrun-common-2.0.0-0.20.rc8.fc35.x86_64.rpm https://kojipkgs.fedoraproject.org//work/tasks/6157/78956157/plater-2.0.0-0.20.rc8.fc35.noarch.rpm https://kojipkgs.fedoraproject.org//work/tasks/6157/78956157/pronsole-2.0.0-0.20.rc8.fc35.noarch.rpm https://kojipkgs.fedoraproject.org//work/tasks/6157/78956157/pronterface-2.0.0-0.20.rc8.fc35.noarch.rpm
```

I'Ve player with that a bit and it seem fine, but I was not printing.

kapanala, 2021-11-16:
Upon use I'm getting type errors reported in the RH screen:

TypeError: Point(): arguments did not match any overloaded call:
  overload 1: too many arguments
  overload 2: argument 1 has unexpected type 'float'
  overload 3: argument 1 has unexpected type 'float'
  overload 4: argument 1 has unexpected type 'float'
Traceback (most recent call last):
  File "/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", line 111, in onPaint
    self.draw(dc, w, h)
  File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", line 289, in draw
    self.highlightQuadrant(gc, self.quadrant, self.concentric)
  File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", line 229, in highlightQuadrant
    self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - inner_ring_radius, a1 + fudge, a2 - fudge)
  File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", line 195, in drawPartialPie
    p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * math.sin(angle1))

kapanala, 2021-11-17:
I am getting type errors in the RH window.   I've put the details into 
the bug report.  I am having email problems, so you may be getting 
multiple messages (or none at all).

kapanala

kapanala, 2021-11-17:
Pronterface starts up, but upon use I'm getting type errors in the RH 
screen:

TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'
Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))

Kapanala

kapanala, 2021-11-17:
Email working again!

This is the type error I get; in the bug report I may have copied across 
two instances.  Difficult to say what action triggers the error.

Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))
TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'

kapanala

kapanala, 2021-11-17:
Further experimentation shows it is related to moving the mouse over the 
x/y/z position control area, which eventually died leaving a blank space.

kapanala

kapanala, 2021-11-17:
Miro,

Please see posted comments.  I am having difficulty getting email to 
reflect back to me, so you may have received several replies.

kapanala

kapanala, 2021-11-17:
Seems to work!  I haven't done a check for all functions.

Thanks,

hroncok, 2021-11-17:
A new one:

```
sudo dnf --setopt=install_weak_deps=False install https://kojipkgs.fedoraproject.org//work/tasks/5013/78985013/plater-2.0.0-0.20.rc8.fc35.noarch.rpm https://kojipkgs.fedoraproject.org//work/tasks/5013/78985013/printrun-2.0.0-0.20.rc8.fc35.x86_64.rpm https://kojipkgs.fedoraproject.org//work/tasks/5013/78985013/printrun-common-2.0.0-0.20.rc8.fc35.x86_64.rpm https://kojipkgs.fedoraproject.org//work/tasks/5013/78985013/pronsole-2.0.0-0.20.rc8.fc35.noarch.rpm https://kojipkgs.fedoraproject.org//work/tasks/5013/78985013/pronterface-2.0.0-0.20.rc8.fc35.noarch.rpm
```

hroncok, 2021-11-17:
You might need to use `reinstall` instead of `install` in that command in case you have the packages from the previous instruction still installed.

hroncok, 2021-11-17:
Could you please paste the terminal output (even if it looks identical)?

kapanala, 2021-11-17:
Pressed connect (pronterface has to be connected to printer for xyz 
control to be active).  Then pass the mouse over the xyz control (no 
clicking).  This is the entire RH screen content.

Connecting...
Printer is now online.
Using tool 0.
Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))
TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'
Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))
TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'
Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))
TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'
Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))
TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'
Traceback (most recent call last):
   File 
"/usr/lib64/python3.10/site-packages/printrun/gui/bufferedcanvas.py", 
line 111, in onPaint
     self.draw(dc, w, h)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 289, in draw
     self.highlightQuadrant(gc, self.quadrant, self.concentric)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 229, in highlightQuadrant
     self.drawPartialPie(gc, center, r1 - inner_ring_radius, r2 - 
inner_ring_radius, a1 + fudge, a2 - fudge)
   File "/usr/lib64/python3.10/site-packages/printrun/gui/xybuttons.py", 
line 195, in drawPartialPie
     p1 = wx.Point(center.x + r1 * math.cos(angle1), center.y + r1 * 
math.sin(angle1))
TypeError: Point(): arguments did not match any overloaded call:
   overload 1: too many arguments
   overload 2: argument 1 has unexpected type 'float'
   overload 3: argument 1 has unexpected type 'float'
   overload 4: argument 1 has unexpected type 'float'

hroncok, 2021-11-17:
Oh, this is a cache issue! Your dnf is not re-downloading the packages but it is reusing the old ones. To workaround, please:

Download https://kojipkgs.fedoraproject.org//work/tasks/5013/78985013/printrun-common-2.0.0-0.20.rc8.fc35.x86_64.rpm e.g. by your web browser.

Run `sudo dnf --setopt=install_weak_deps=False reinstall ~/Downloads/printrun-common-2.0.0-0.20.rc8.fc35.x86_64.rpm` (the last argument is a path to the downloaded RPM, it might differ on your machine.

kapanala, 2021-11-17:
Now it works!  Thanks!  Just one minor issue, when I start pronterface 
from the command line, the terminal displays multiple GTK errors.  I 
believe this has been an issue for some time.

I had a previous version of this email disappear; this may be a 
duplicate copy.

kapanala

$ pronterface

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.275: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.275: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.276: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.276: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.276: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.276: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.334: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.334: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.334: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.334: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.334: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.334: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.410: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.410: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.411: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.411: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.411: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.411: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.440: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:18.440: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
[... 167 lines cut ...]
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.494: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.495: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.523: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.523: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.523: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.523: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.523: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface:27375): Gtk-CRITICAL **: 11:11:26.523: 
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

closed 2021-11-18 by hroncok