kliment/Printrun #1226: Please help me to use ESP3d with my old printer

enghady-ahmed, 2021-11-15:
i use ESP3D to connect to my old printer through Wifi i success to connect but unfortunately i can't print from SD , although i can print using pronterface ,so can you help me with Gcode sequences that can i use manual to start print Gfile from SD like
1-M21 to int SD
2- M20 to list file
3- M23 "File name" to open it
4-M24 to start printing

i try the above , but it didn't work i try to check pronterface py file to know the code but i get confused

here logs i get when i connect 
Connecting...
Sprinter
1.3.22T / 20.08.2012
start
Printer is now online.
SD Start
Card type: SD2
Volume type is FAT16
Volume size (Mbytes): 1873
Soft PWM Init
Planner Init
Stepper Timer init
Free Ram: 595
Plan Buffer Size:1152 / 16

rockstorm101, 2022-01-29:
Hi @enghady-ahmed, sorry I missed your post here. Did you find out how to make it work already? It might be useful to share your experience here for future users. Thanks

enghady-ahmed, 2022-01-30:
no and i planned to upgrade to Arduino and use Marlin