kliment/Printrun #1229: Printrun cannot be run in comercial environments and is non-free

MorganJamesSmith, 2021-11-25:
So there is a file printrun/packer.py that relies on polygon (https://www.j-raedler.de/projects/polygon/). Polygon has a c component shipped with it called GPC. GPC cannot be used in commercial environments and because of that it is non-free software.

I'm not sure what packer.py does or how deeply polygon is integrated into it, but I do hope we could not use this software.

I understand this might seem quite silly to some people and it also seems silly to me as well.  However, it would be nice to be able to use this software in a commercial environment or be able to distribute it on fully free distros.

kliment, 2021-11-25:
I just had a look at this. It seems it was originally introduced by @iXce back in 2013. The code that calls it is [here](https://github.com/kliment/Printrun/blob/master/printrun/objectplater.py#L152) - it attempts to import packer.py, which fails if Polygon is not installed on the system, and uses it only if available. If not, it falls back to another algorithm. We neither distribute nor require Polygon to be installed. You can also install any other API-compatible Polygon implementation that doesn't depend on gpc and that would import fine. By default, the import is going to fail and Polygon will not be used at all unless you explicitly choose to install it. None of this makes Printrun non-free or not useable in commercial environments. I appreciate your concern but in this case it's not justified.

closed 2021-11-25 by kliment

MorganJamesSmith, 2021-11-25:
That's good the hear. Thank you so much for letting me know!

I'm trying to package this for GNU Guix and was wondering how I can get the unit tests to pass without polygon.

I'm running `python setup.py test` and I get this error

```
======================================================================
ERROR: packer (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: packer
Traceback (most recent call last):
  File "/gnu/store/9w9jvy3bgjg4qaqmrij01nbppiccqr7c-python-3.8.2/lib/python3.8/unittest/loader.py", line 154, in loadTestsFromName
    module = __import__(module_name)
  File "/tmp/guix-build-printrun-2.0.0rc8.drv-0/source/printrun/packer.py", line 22, in <module>
    import Polygon
ModuleNotFoundError: No module named 'Polygon'
```

kliment, 2021-11-25:
packer.py fails to import if Polygon is not present. It's our internal wrapper for the Polygon functionality, which obviously doesn't work without Polygon being present. Therefore its unit test will always fail in that case. On a system where Polygon will never exist that unit test makes no sense and you can remove it. If you want, you can also remove the entire packer.py file, since it serves no purpose when Polygon is not present.

MorganJamesSmith, 2021-11-25:
Deleting the file worked! Package complete! It'll be in Guix soon. Thanks!

kliment, 2021-11-25:
Thank you for your help in packaging this!

VinnyCordeiro, 2021-11-25:
> Deleting the file worked! Package complete! It'll be in Guix soon. Thanks!

I believe it would be useful to add a note about that somewhere on the Guix package, just for the sake of completeness.