kliment/Printrun #1230: Pronterface wont run because of assertion error.

Pruchol5, 2021-12-07:
Hi i am having a lot of trouble getting Pronterface to work. The Exe straight up refuses to launch with a "Failed to execute script Pronterface".

Eventually, I tried to build it myself. With a lot of trouble it seemed to work but now when I try to launch the built Exe I get the following error. (This error also occurs when I try to launch it with: "python pronterface.py" 

Traceback (most recent call last):
  File "pronterface.py", line 62, in <module>
    app = PronterApp(False)
  File "X:\Gejms\Printrun-printrun-2.0.0rc8\printrun\pronterface.py", line 2455, in __init__
    self.mainwindow = PronterWindow(self)
  File "X:\Gejms\Printrun-printrun-2.0.0rc8\printrun\pronterface.py", line 186, in __init__
    self.SetIcon(wx.Icon(iconfile("pronterface.png"), wx.BITMAP_TYPE_PNG))
wx._core.wxAssertionError: C++ assertion "strcmp(setlocale(LC_ALL, NULL), "C") == 0" failed at ..\..\src\common\intl.cpp(1579) in wxLocale::GetInfo(): You probably called setlocale() directly instead of using wxLocale and now there is a mismatch between C/C++ and Windows locale.
Things are going to break, please only change locale by creating wxLocale objects to avoid this!

Hope you guys can help. I don't want to spend 50 bucks on a Raspberry Pi :D

kliment, 2021-12-08:
@DivingDuck do you have an idea what this could be? Looks like it's something going wrong in iconfile

DivingDuck, 2021-12-08:
Hi @kliment,
My guess is more on wxPhthon. I'll check with Pruchol5 how he did the compiling.

@Pruchol5, 
What Python- and wxPython version was used for building the exe. In addition how did you build your exe? Your OS is Windows 10 / 11 x64?

DivingDuck, 2021-12-16:
@Pruchol5, 
any news?
Best regards,
DD

Pruchol5, 2021-12-16:
@DivingDuck 
Sorry forgot about this. Yes I am running windows 10 x64. I tried building with the newest version of python and the recommended 3.0.7 version.

Tried building with both wxpython 4.0.7? And 4.1.1.
I built it with the batch file. This downloaded all the reps and built the file. Before doing this I ran the upgrade commands for both the apt and build tools. In the meantime I tried some other things like downloading the .net build tools (the batch file gave this as an error while trying to download wxpython) and some other things I can't remember.

DivingDuck, 2021-12-16:
Both version of wxpython won't work. You need to use the latest development version. In addition when you run py >= 3.9 you need to adjust the requirements.txt for wxpython to >=4.1.1 as lower version will crash with a compiling error.

I have an updated version of [release_windows.bat](https://github.com/DivingDuck/Printrun/blob/test/release_windows.bat) in my test branch. Copy and replace that file in your local source code directory and follow the instructions in the beginning of the batch file. Running the batch will compile you a working executable file. It works for cmd promt and PowerShell.

Let me know if this works for you.

DivingDuck, 2021-12-23:
@Pruchol5,
Had that work for you?
Best regards, 
DD

poikilos, 2022-08-25:
When researching an issue I found this wxPython issue with the same error in Phoenix (Phoenix is the official rewrite of wxPython, but I'm unsure if you're using Phoenix): https://github.com/wxWidgets/Phoenix/issues/769

DivingDuck, 2023-02-12:
@Pruchol5,

I know it was long time ago you opened this issue. We got finally a fix for this issue.

You can test and give us a feedback if you like so. Here is a link to the latest Windows build.
https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark as completed

closed 2023-02-12 by DivingDuck