kliment/Printrun #1231: Acknowledgement for creation of the Printrun/Pronterface logo.

ahmetcemturan, 2021-12-08:
Hi,
I would like to ask You to put an acknowledgement on the creator of the logo (me).
I noticed that Alessandro from slic3r did that and think that it would be nice to get some recognition.
You can confirm that with Kliment.

Best Regards.

kliment, 2021-12-08:
Hello Ahmet, good to hear from you again! I would definitely be happy to acknowledge you as the author of the logo. Where would you like us to mention it?

ahmetcemturan, 2021-12-09:
Hi Kliment,

 

What are You up to?

 

Regarding Acknowledgement:

 

That’s up to You.  On the website or in the code. Or both.

It’s just that I have started to produce Filament now and I would like people to know that I am not just in it for the money but a grassroots Reprapper.

So I would mention that and put a link etc.

 

We have produced some rel good quality filament.  I would say better than Josef Prusa’s 😊

 

I can send You a roll or two if You want.  We will soon sell also in Europe.

 

It’s brand is Mia Filament:  www.miafilament.com

 

Best Regards

Ahmet

kliment, 2022-03-26:
Hey @ahmetcemturan, sorry it took so long but I've now added an acknowledgement for everyone who contributed to Printrun so far, including you. See https://github.com/kliment/Printrun/blob/master/CONTRIBUTORS.md - you're listed as both developer and artist.

closed 2022-03-26 by kliment

ahmetcemturan, 2022-05-19:
Thank You very much Kliment.

 

I mean it’s one of these:” not really necessary but nice to have “ things.

commit 71cdf0ba (2022-03-26): Credit all contributors in about dialog. Fixes #1231