kliment/Printrun #1233: PID Tune Error SKR MINI E3 V2

DarthCookies, 2021-12-28:
Hey Guys,

i tried everything i know, switching usb cabels and what not, but my PID Tune will constantly fail. 

thats what my debug says:
echo: PID_DEBUG 0: Input 44.08 Output 0.00 pTerm 0.00 iTerm 0.00 dTerm 0.00

and that is the error  i get: 

>>> m303 s210 e0 c8
SENDING:M303 S210 E0 C8
PID Autotune start
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Das Gerät erkennt den Befehl nicht.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range
Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "printrun\printcore.py", line 744, in _send
  File "serial\serialwin32.py", line 317, in write
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'Das Gerät erkennt den Befehl nicht.', None, 22))

any tips on what i can look for?
PID is enabled in configure.h

volconst, 2021-12-28:
Duplicate of #1204 which seems still unresolved.

DivingDuck, 2021-12-28:
What version of Printrun is running?

In the past we had this issue https://github.com/kliment/Printrun/issues/1204 with the same SKR board but this issue is still open.

Unfortunately I do not have this board so I can't really help you. On my board with smoothieware sending M303 E0 S245 works just fine.

Errors with serial connection can point as well to a bad USB connection / too fast baud rates / too long USB cable. Maybe worth to check too.

DivingDuck, 2021-12-28:
Hi @volconst you was a little bit faster... :)

volconst, 2021-12-28:
Hi, @DivingDuck 
I was tempted to tell you the same, but you win more points for details. If I knew you will answer, I would give you way.
Cheers :)

DarthCookies, 2021-12-28:
im using 2.0.0rc8 on win 10.
for now i will try to pid via arduino or via gcode file or use the PID on my printer itself.  i will answer if that worked for me

rockstorm101, 2022-02-23:
Closing. Duplicate of #1204.

closed 2022-02-23 by rockstorm101