kliment/Printrun #1234: Temperatures not displaying in Pronterface

SidSkiba, 2022-01-11:
I am using Pronterface 2.0.0rc8 X64.
I have a Marlin setup with V2.0.9.2.
It is a 2 extruder printer.
The temp gauges nor the graph report he printer temperatures. They simply display 0C. They do show the setpoints issued to printer fine. Just not the actual temps.

The monitor window shows regular temp values:
ok T0:24.06 /0.00 B:23.45 /0.00 T0:24.06 /0.00 T1:24.28 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:24.00 /0.00 B:23.53 /0.00 T0:24.00 /0.00 T1:24.16 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:24.00 /0.00 B:23.50 /0.00 T0:24.00 /0.00 T1:24.14 /0.00 @:0 B@:0 @0:0 @1:0

When I enable debug communications I get:
SENT: M105
RECV: ok T0:24.00 /0.00 B:23.48 /0.00 T0:24.00 /0.00 T1:24.08 /0.00 @:0 B@:0 @0:0 @1:0
SENT: M105
RECV: ok T0:24.00 /0.00 B:23.50 /0.00 T0:24.00 /0.00 T1:24.03 /0.00 @:0 B@:0 @0:0 @1:0
SENT: M105
RECV: ok T0:24.00 /0.00 B:23.45 /0.00 T0:24.00 /0.00 T1:24.06 /0.00 @:0 B@:0 @0:0 @1:0

The printer LCD is reporting 23-24C for Bed and Hotends (not printing - idle)

If I do heat the E0 and Bed I do so values changing in monitor window.
SENT: M104 S200.0
RECV: ok
Setting bed temperature to 60 degrees Celsius.
SENT: M140 S60.0
RECV: ok
SENT: M105
RECV: ok T0:27.24 /200.00 B:23.50 /60.00 T0:27.24 /200.00 T1:24.00 /0.00 @:127 B@:127 @0:127 @1:0
SENT: M105
RECV: ok T0:126.26 /200.00 B:28.00 /60.00 T0:126.26 /200.00 T1:24.33 /0.00 @:127 B@:127 @0:127 @1:0

Regards,

Sid

DivingDuck, 2022-01-11:
Did you check the diagram in -->Settings -->User interface -->Display temperature graph and gauges? 

[image]

There is maybe something wrong with your firmware setup / reporting temps:

```
         T0:24.06 /0.00 B:23.45 /0.00 T0:24.06 /0.00 T1:24.28 /0.00 @:0 B@:0 @0:0 @1:0
RECV: ok T0:126.26 /200.00 B:28.00 /60.00 T0:126.26 /200.00 T1:24.33 /0.00 @:127 B@:127 @0:127 @1:0
```

T0, then B, then T0 once more, then T1 then all the @ @ @ @. Two times T0 in one reporting line, is this normal?

This is how it looks like for two extruders and 2 beds on my printer (firmware is Smoothieware):

[image]

```
         T:39.9 /180.0 @255 T1:20.7 /180.0 @255 B:23.3 /30.0 @255 B1:22.4 /30.0 @255
RECV: ok T:39.9 /180.0 @255 T1:20.7 /180.0 @255 B:23.3 /30.0 @255 B1:22.4 /30.0 @255
```

SidSkiba, 2022-01-12:
Hi DivingDuck, Thank you for the reply. I checked and yes the boxes are checked:  And they are showing on the interface. Just no values from the printer. It is very possible that Marlin has an issue that has not been noted yet. An old firmware was working with old Pronterface.  I do not know what the correct format of the M105 is supposed to be. It does differ from yours. I will ping the Marlin team if they can advise. Sid Sent from Mail for Windows From: DivingDuckSent: January 11, 2022 2:41 AMTo: kliment/PrintrunCc: SidSkiba; AuthorSubject: Re: [kliment/Printrun] Temperatures not displaying in Pronterface (Issue #1234) Did you check the diagram in -->Settings -->User interface -->Display temperature graph and gauges?There is maybe something wrong with your firmware setup / reporting temps:         T0:24.06 /0.00 B:23.45 /0.00 T0:24.06 /0.00 T1:24.28 /0.00 @:0 B@:0 @0:0 @1:0RECV: ok T0:126.26 /200.00 B:28.00 /60.00 T0:126.26 /200.00 T1:24.33 /0.00 @:127 B@:127 @0:127 @1:0T0, then B, then T0 once more, then T1 then all the @ @ @ @. Two times T0 in one reporting line, is this normal?This is how it looks like for two extruders and 2 beds on my printer (firmware is Smoothieware):         T:39.9 /180.0 @255 T1:20.7 /180.0 @255 B:23.3 /30.0 @255 B1:22.4 /30.0 @255RECV: ok T:39.9 /180.0 @255 T1:20.7 /180.0 @255 B:23.3 /30.0 @255 B1:22.4 /30.0 @255

SidSkiba, 2022-01-12:
It has been confirmed a bug in Marlin. It has been fixed now in the bugfix branch. It all works good now. 

https://github.com/MarlinFirmware/Marlin/pull/23505

Thank you.

DivingDuck, 2022-01-12:
Good to know and thanks for your update. I will close the issue as your problem is solved.
Best regards, DD

volconst, 2022-01-12:
happy end, but pronterface is wrong here - it only accepts 'T:' and not 'T0:'
https://github.com/kliment/Printrun/blob/faade5b9f33779a9b08286498a15eefd0abd13e0/printrun/printcore.py#L432
The B@:0, @0:0 notation that looked as garbage to me is explained at 
https://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature

DivingDuck, 2022-01-12:
Hi @volconst ,
Interesting, I had only a quick look on an other place (my bad):
https://github.com/kliment/Printrun/blob/faade5b9f33779a9b08286498a15eefd0abd13e0/printrun/pronterface.py#L1865-L1872
and thought this is consequently done on other places too.

Yeah, I had seen the explanation from reprap too and was as well a bit irritated about the logic to split values that should be reported together.

So, what do you think, should printcore.py be updated so it can handle both T and T0?

volconst, 2022-01-12:
@DivingDuck ,
My bad, you are on the right location.
I still think that both 'T' and 'T0' should work and do not understand why they don't.
I am not at my dev computer, but tried the regex and it parsed the temps and this was my second guess.
My location seems to be for callbacks that do not handle temperature graph. They may need fix also if someone uses them.

SidSkiba, 2022-01-13:
> @SidSkiba
> Can you test if this branch https://github.com/volconst/Printrun/tree/empty-temp-graph works with your printer?
> I saw that the problem is fixed in marlin, but in my opinion the fix belongs to pronterface.
> The branch should work with unpatched marlin, but if updating (downgrading) firmware is too much it is also fine to test on the patched firmware that at least it does not break other printers.

Will test tonight

SidSkiba, 2022-01-14:
> @SidSkiba Can you test if this branch https://github.com/volconst/Printrun/tree/empty-temp-graph works with your printer? I saw that the problem is fixed in marlin, but in my opinion the fix belongs [...]

Is there a precompiled version of the branch? Has the version linked here been updated?
https://github.com/kliment/Printrun/releases

I had been downloading the X64 precompiled version.

volconst, 2022-01-16:
@SidSkiba 
I added in my branch https://github.com/volconst/Printrun/tree/empty-temp-graph
a new commit https://github.com/volconst/Printrun/commit/bcbfec6c850e72a38a774a9176c366e8f6c1da4c
which shows the temps of the selected tool in the temperature gauge.
Please test it. I refactored the code, so it may contain regressions :)
See testtools/mock-printer.py in the same branch, you can change the temperature status line sent from the printer, needs restart of mock-printer.py

SidSkiba, 2022-01-17:
I tried it and the gauge displays Heater 0. However when I change tool on screen I get this:

Setting hotend temperature to 100 degrees Celsius.
Traceback (most recent call last):
  File "printrun\gui\controls.py", line 344, in extrudersel_cb
  File "printrun\pronterface.py", line 661, in tool_change
AttributeError: 'Button' object has no attribute 'Value'
Traceback (most recent call last):
  File "printrun\gui\controls.py", line 344, in extrudersel_cb
  File "printrun\pronterface.py", line 661, in tool_change
AttributeError: 'Button' object has no attribute 'Value'
Traceback (most recent call last):
  File "printrun\gui\controls.py", line 344, in extrudersel_cb
  File "printrun\pronterface.py", line 661, in tool_change
AttributeError: 'Button' object has no attribute 'Value'
Traceback (most recent call last):
  File "printrun\gui\controls.py", line 344, in extrudersel_cb
  File "printrun\pronterface.py", line 661, in tool_change
AttributeError: 'Button' object has no attribute 'Value'
Setting hotend temperature to 0 degrees Celsius.
Traceback (most recent call last):
  File "printrun\gui\controls.py", line 344, in extrudersel_cb
  File "printrun\pronterface.py", line 661, in tool_change
AttributeError: 'Button' object has no attribute 'Value'
Traceback (most recent call last):
  File "printrun\gui\controls.py", line 344, in extrudersel_cb
  File "printrun\pronterface.py", line 661, in tool_change
AttributeError: 'Button' object has no attribute 'Value'

And the gauge does not change...continues to show Heater 0.

Also it seems I cannot set the temp on Tool 1. It just changes Tool 0.

volconst, 2022-01-17:
@SidSkiba 
Oops, 2 extruders are handled with button, more - with drop down :)
I added fix for this in https://github.com/volconst/Printrun/tree/empty-temp-graph

SidSkiba, 2022-01-17:
Looks like it works. The gauge changes with the tool selection and shows the correct temps for the selected tool.

Thank you for the awesome work.

commit db31aed2 (2022-01-18): Fix empty temperature graph
Fixes #1234
closed 2022-01-18 by volconst