kliment/Printrun #1235: Use Macro for live auto change GCODE command

ElfyShort, 2022-01-12:
Hello,

I would like to interrupt a gcode like "M104 S500" by "M104", extract an "S" number (500 in this case), do math with it (separate it in bits - % and / by 255) and send as another batch of commands instead, for example as instruction to the I2C interface:

M260 A99  ; Target slave address
M260 B1  ; 
M260 B245 ; 
M260 S1   ; Send the current buffer

Is it possible to do using a macro tool?
Thank you!

volconst, 2022-01-12:
You have a gcode file containing a line like "M104 S500" and you want it replaced with several other lines.
It can be done with macro, but a much cleaner implementation is to do it as text processing with any language of choice outside of pronterface.

ElfyShort, 2022-01-13:
> You have a gcode file containing a line like "M104 S500" and you want it replaced with several other lines. It can be done with macro, but a much cleaner implementation is to do it as text processing with any language of choice outside of pronterface.

Thank you for the insight. 
The solution we use now is exactly that. Unfortunately this introduce an additional processing step that is an inconvenience. 
Could you give any hints on how do I interrupt a command with a floating number? In my GCODE "M104 Sx" "x" will always vary and I need to get that number to construct a new command before exchanging to new command.

volconst, 2022-01-13:
Create a macro with this contents:
```
!import re
!i = 0
!for li, layer in enumerate(self.fgcode.all_layers):
!  new_layer = []
!  for lni, ln in enumerate(layer):
!    #print('li %d lni %d'%(li, lni))
!    if ln.command == 'M104':
!      m = re.search(' +S([\d\.d]+)', ln.raw)
!      if m:
!        tmp = int(m.group(1))
!        print(tmp)
!        new_layer.extend(['M260 A99', 'M260 B%d'%(tmp//256), 'M260 B%d'%(tmp%256), 'M260 S1'])
!        i += 1
!      else: new_layer.append(ln.raw)
!    else: new_layer.append(ln.raw)
!  if len(new_layer) != len(layer):
!    self.fgcode.rewrite_layer(new_layer, li)
!print('replaced %d commands'%i)
```
You can also create a button invoking the macro.
Call the macro after a file is loaded and then you can print the file.
The macro should be run once on a file.

ElfyShort, 2022-01-14:
> 

Wow!
Thank you very much, volconst!

I have one more question. 
Lets say I will name this macro "WOW", and then add a line in the beginning of my GCODE :
WOW;

Will this call the macro automatically? 

As I understood, It should be called when the name of the macro is typed in and send in the text field, and the custom button does exactly that (spare you from typing the name). Is this correct?

volconst, 2022-01-14:
My experience is that macros are not executed if seen in gcode files.
Even if they were, the macro modifies an already loaded file - so it should be at the end of the file.

> As I understood, It should be called when the name of the macro is typed in and send in the text field, and the custom button does exactly that (spare you from typing the name). Is this correct?
Yes

volconst, 2022-01-16:
@ElfyShort 
You can execute a macro from a file with the host command syntax
```
;@WOW
```

ElfyShort, 2022-01-17:
Thank you for the information and extensive support!

closed 2022-01-17 by ElfyShort