kliment/Printrun #1236: Custombtn

bsjoelund, 2022-01-17:
Pronterface will not start when custombtn.txt is present.  Debian 10

rockstorm101, 2022-01-18:
Hi @bsjoelund I'm unable to reproduce this. I assume you are running from source. What version of Printrun are you using? Or do you use de Debian package? Could you please share the error you get when you launch Pronterface from a terminal?

bsjoelund, 2022-01-18:
Hi,

Debian package

1.6:0

Cheers
Bengt

Den 2022-01-18 kl. 07:38, skrev Rock Storm:

>
> Hi @bsjoelund <https://github.com/bsjoelund> I'm unable to reproduce 
> this. I assume you are running from source. What version of Printrun 
> are you using? Or do you use de Debian package? Could you please share 
> the error you get when you launch Pronterface from a terminal?
>
> or Android 
***@***.***:~$ pronterface
WARNING:root:Could not setup DBus for sleep inhibition: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not provided by any .service files
Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 66, in <module>
    app = PronterApp(False)
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 2296, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 220, in __init__
    self.reload_ui()
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 282, in reload_ui
    self.settings.controlsmode == "Mini")
  File "/usr/lib/python2.7/dist-packages/printrun/gui/__init__.py", line 294, in createGui
    self.cbuttons_reload()
  File "/usr/lib/python2.7/dist-packages/printrun/pronterface.py", line 1883, in cbuttons_reload
    b = wx.Button(self.centerpanel, -1, btndef.label, style = wx.BU_EXACTFIT)
AttributeError: 'list' object has no attribute 'label'

btns=[
###Defining custom buttons for pronterface is easy. Here's how.
###Below these instructions, add a line with the following format for each button
# ["button name","command",color(RGB)], <--That comma is important, do not forget it
###As an example: (Remove the # to try it out):

###You can use gcodes or any pronsole/pronterface commands
###The first four buttons will end up at the top of the window, visible in mini mode
###The rest of the buttons will be at the bottom under the gcode preview
###ADD BUTTON DEFINITIONS BELOW THIS LINE
["Read temp","M105",(200,100,100)],
["Disable Axes","M84",(400,100,100)],

###ADD BUTTON DEFINITIONS ABOVE THIS LINE
]

rockstorm101, 2022-01-18:
Could it be that you messed up the configuration file? If you try to generate a new configuration file, does that make it work? For instance saving a copy of the config file like:
```bash
mv ~/.pronsolerc ~/.bak_pronsolerc
```

rockstorm101, 2022-03-05:
Hi @bsjoelund, any luck with the suggested solution?

DivingDuck, 2025-02-15:
@rockstorm101,

we should close this issue. Since long time we do not use custombtn.txt anymore as it was transferred into the config file. Python 2.x is also not supported quite long.

There is a left over in our source code for transferring the custombtn.txt definitions into the config file where it is maybe time to get rid off too (mentioned in #1286), but this should be independent from the issue here.

neofelis2X, 2025-02-15:
> There is a left over in our source code for transferring the custombtn.txt definitions into the config file where it is maybe time to get rid off too
 
Yes that code is there and it is currently broken.  It just causes an error when it encounters a custombtn.txt. I fixed it in one of my branches somewhere, but I never came around to push it.

rockstorm101, 2025-02-16:
Funny you mention these. I was working on some refactoring recently and stumbled upon this issue as well. I'll submit the refactoring for review and remove the custombtn code completely (or open up the discussion for it). That shall close these two issues. Thanks for bringing these up, I didn't have them on my radar.

PR #1483 (2025-02-17): refactor: Remove legacy custombtn and configfile
This PR removes code, which is not standard practice and constitutes a breaking change. Such changes should only be introduced on major version upgrades. However I believe it could be justified in this case and only a minor version upgrade would be needed in the future.

custombtn.txt was deprecated before Printrun 2.x but the code was left. The code didn't even work as expected so this potentially breaking change has a very low chance of affecting any users.
Closes #1236

The `configfile` function was outdated and only used for handling customb [... cut]
commit 1af6ad91 (2025-02-17): refactor: Remove legacy custombtn and configfile (#1483)

custombtn.txt was deprecated before Printrun 2.x but the code was left. The
code didn't even work as expected so this potentially breaking change has a
very low chance of affecting any users.
Closes #1236

The `configfile` function was outdated and only used for handling
custombtn.txt so it is safe to remove as well.
Closes #1286
commit 6a57b2f8 (2025-03-08): refactor: Remove legacy custombtn and configfile

custombtn.txt was deprecated before Printrun 2.x but the code was left. The
code didn't even work as expected so this potentially breaking change has a
very low chance of affecting any users.
Closes #1236

The `configfile` function was outdated and only used for handling
custombtn.txt so it is safe to remove as well.
Closes #1286
closed 2025-02-17 by rockstorm101