kliment/Printrun #1239: Rendering on the viewer

daveysaxt, 2022-01-24:
I'm not sure if this is a feature request or a bug report...

Have a look at my shot of something on the viewer (oh, and yes, it's a restarted long long long print, but this issue occurs for all models complete or not):

[image]

The model is rendered wider at the 'back' than the front. Has anyone else actually noticed this? I realised this after quite a long time of 'feeling' something just wasn't quite right.

Shouldn't the model be rendered wider at the front and narrower at the back to achieve correct perspective ?

Maybe I'm being pedantic , but it looks really weird to me (also, with a complete model rendered, that optical illusion where the thing turns inside-out and nearly boils ya brain can be experienced more often with the confusing perspective) and when rotating/translating the view using the mouse it 'feels' kind of physically awkward because the perspective is confusing.

In all, it's  kind of 'uncomfortable' - if you get what I mean ?

Please can this be changed to correct the perspective and possibly a toggle added in settings to select which perspective mode to do ?

kliment, 2022-01-24:
I see what you mean - we don't actually use a perspective transform, so you're getting an isometric view - it should be straight when viewed exactly from the front, and will look weird from most angles. Enabling perspective transform is easy enough (I'll make a pull request to do that now so you can test) but it makes the 3d navigation a bit confusing.

kliment, 2022-01-24:
@daveysaxt this functionality is now merged into the master branch, if you are running from source you can try it out

closed 2022-01-24 by kliment

daveysaxt, 2022-01-26:
> @daveysaxt this functionality is now merged into the master branch, if you are running from source you can try it out

@kliment Okidoke... sorry for late reply, I've been very busy with family issues last couple of days. I did start setting up for running from source but ran into issues with it, but going to start afresh with python removal and reinstall. Just a quick question: Should it work with python 3.10, or does it have to be python 3.6 ?

daveysaxt, 2022-01-26:
@kliment So, got the run from source env set up - there's a compatibility issue that stops wxPython building with 3.10, so just started afresh with 3.6.8. Have tried the perspective option. It's Great! Thanks very much for that.

PR #1240 (2022-01-24): Add a perspective transform option, render gcode views with perspective transform
When the perspective option in settings->options->viewer is enabled, renders the gcode viewer with a perspective transform instead of an isometric transform. Fixes #1239