kliment/Printrun #1241: Pronsole closes after losing connection

dVaGaymer, 2022-01-31:
So I have a raspberry connected to the printer. Ssh is enabled so when I'm not home I connect to the raspberry, init pronsole and start printing, but as soon as I terminate the ssh session or lose connection the printer stop printing and and can't resume it.
Is there any way to keep pronsole running, or keep the printer printing even though pronsole closes?
Thank you, people 😄️

kliment, 2022-01-31:
You might want to look into tmux or gnu screen for this.

dVaGaymer, 2022-01-31:
What does tmux have to do with this?
What i mean is that when I'm connected to the raspberry and lose the Internet connection (which happens a lot because my isp is not good) the ssh server terminates and so does the printer, can't it continue printing even when pronsole closes?

kliment, 2022-01-31:
When you have tmux running on the pi, you can keep a persistent session that keeps running even if ssh terminates, and then you can reconnect to it. Check out the tmux documentation.

dVaGaymer, 2022-01-31:
Ah okay, I got it now, I will, thank you

closed 2022-01-31 by kliment