kliment/Printrun #1242: PRONTERFACE DROPPED DUE TO PRINTRUN FAIL

mtnhawker, 2022-02-09:
BIGTREE SKR2 VB RUNNING MARLIN 2.0 W PRONTERFACE 

 I SEE THE FOLLOWING MESSAGE AFTER A MINUTE OR SO CONNECTION AND MESSAGE PRINTER IS ONLINE AND THIS ERROR ERROR REPEATS OVER AND OVER

 Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "printrun\printcore.py", line 420, in _listen
  File "printrun\printcore.py", line 354, in _readline
IndexError: tuple index out of range
Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "printrun\printcore.py", line 744, in _send
  File "serial\serialwin32.py", line 317, in write
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

rockstorm101, 2022-02-23:
Hi @mtnhawker, did you have any luck fixing this already? Otherwise, could you provide more information regarding how you get to this error? What version of Printrun are you using? I assume you run a Windows machine? Are you using pronsole or pronterface?

mtnhawker, 2022-03-01:
I have not resolved my issues with the Bigtree SKR V2 using the TFT 3.5
V3.0. With the TFt attached the SKR will reset every few minutes making it
unusable. It also resets randomly without the TFT display plugged in . It
reset with the micro memory card in or out and it reset even with just the
PC into the USB port at 250000 using com5. I searched the web and others
have reported similar issues.
With just the PC (no card and no TFT) it printed for 2 hours before it
reset. I have 2209 motor drives from Bigtree for this board.  I sure don't
trust this setup enough to use it and I am not sure how to fix it.
What do you think?
thanks
brian Lundy
SLC UTAH

volconst, 2022-03-03:
Please try the exe from latest branch commit at the bottom of https://github.com/kliment/Printrun/actions/runs/1742349606
It contains some fixes, though it may fail in a different way.
Watch the LCD for signs whether the printer it self restarts.
We need to determine whether the communication reset is caused by printer restart, which is printers problem or the com reset is some recoverable event.
To test whether the printer has restarted you may check whether it knows it's current coordinates, and they are correct, or it stopped heating.

mtnhawker, 2022-03-21:
> Hey, i have this same issue volconst, The screen flashes black and resets when the computer loses connection.

Mine reset from 2 issues:
Baud rate would reset to a low number after downloading firmware and firmware losing bits requiring reinstall (with no changes). Very frustrating but now it works
B

volconst, 2022-03-21:
@gnomea , is the restart reproducible? Does it always happen during the same operation? It is possible that it restarts due to low power, thermal runaway, etc.
Try saving the commands that make it restart to a sd card file and print it without using pronterface. This way we can distinguish whether it is the printer of pronterface problem.

gnomea, 2022-03-22:
the LCD screen flashes the BTT/Marlin restart screen when it disconnects from the computer, can reproduce every time I reconnect a computer after between 1-3 minutes

gnomea, 2022-03-22:
@volconst sorry forgot to tag

volconst, 2022-03-22:
@gnomea 
Send M503 and look for M85 setting.
https://marlinfw.org/docs/gcode/M085.html
Is M85 present?
Try sending a different values. Save with M500.
Does it affect the time it takes to reset?

Escrich, 2022-05-07:
I’m using a BTT Octopus V1.1 board, Marlin version "2.0.9.3", at  115000 bauds,  getting this kind of errors at stopping momment:

Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
  File "printrun\gui\graph.py", line 391, in StartPlotting
RuntimeError: wrapped C/C++ object of type Graph has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 2311, in Notify
  File "wx\core.py", line 3508, in Notify
  File "printrun\gl\panel.py", line 139, in Refresh
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
Traceback (most recent call last):
  File "wx\core.py", line 3383, in <lambda>
RuntimeError: wrapped C/C++ object of type Gviz has been deleted

This makes the printer stops, without errors on it, and it's not possible, in the most of the cases to save the printed part.
I was tried some months ago with a commercial Anet ET4 printer, (stm32 based board), with same results, and it’s currently working with an old Pronterface version, I have not version number here, at same time I’m using every day the last version of Pronterface, with my Original Prusa Mk3S printer, (Einsy Rambo board, Atmega based), without any kind of problem.
But coming back to the Octopus board, STM32 based, I was tried to reduce baud rate till 38400 bauds, activated the whole things included on Marlin to make serial communication more reliable, with same results till now, random stopping at less expected moment, could be in the beginning or near the end of the printing, but not reliable at all.

Escrich, 2022-05-07:
Something positive, I can confirm you, that you can use version 2014.03.10 Without those problems, it was something added after this release

volconst, 2022-05-08:
@Escrich , this is not related to the original error and happens only when you add remove components from the settings dialog. Create a new issue if it happens even after restart of pronterface

brandonbowmansr, 2022-11-19:
Dropping the baud rate to 9800 worked for me, this was with the TFT35 e3 v3.0.1 and skr mini e3 V2 running marlin 2.1.1,  my issue was the same as OP, my proterface was connecting at 112800 before, the rate I have set in marlin, did not check the display speed but 9800 worked fine and I've been running pid for 10 cycles now with no issues, prior it wouldn't make it thru 1.