kliment/Printrun #1243: ERROR:root:Not connected to printer.

ProbablyAnds, 2022-02-22:
Hi there, Ive tried with mutliple different branches of the project to no avail.
I am using a Ardiuno Mega 2560, with a Ramps 1.4.

`from printrun.printcore import printcore
from printrun import gcoder
p=printcore('COM5',250000)

#while not p.online : time.sleep(1)

#If you need to interact with the printer:
p.send_now("G0 x100 y0") # this will send M105 immediately, ahead of the rest of the print
`

I have this in the __init__.py folder and I get this output
ERROR:root:Not connected to printer.

rockstorm101, 2022-02-23:
If (and only if) connection is succesful, it takes a bit of time before the printer comes online and allows sending commands. What happens if you uncomment the following line? (I.e. drop the hash '#' symbol)
```
#while not p.online : time.sleep(1)
```

rockstorm101, 2022-02-23:
> The ardiuno connects perfectly fine with pronterface, but just not through the python code.

Well if Pronterface works a code similar to what you propose above should work. Apart from [allowing the printer to connect][1]. Things that come to mind:

1. Import the time library to be able to sleep: `import time`
2. Double check that "COM5" is the right port to use. It might change from time to time or when other stuff is connected to other ports.
3. Make sure the printer is connected to a power supply capable of moving the motors. Typically, if only connected through the computer, the printer will respond to commands but won't be able to move the motors.
4. Depending on your firmware, the command `G0 x100 y0` might not work straight away. Try sending command `G28` (which homes all axes) first.
```
from printrun.printcore import printcore
from printrun import gcoder
import time

p=printcore('COM5',250000)
while not p.online : time.sleep(1)

p.send_now("G28")  # home all axes, should see some response from the printer by now
p.send_now("G0 x100 y0") 
```

[1]: https://github.com/kliment/Printrun/issues/1243#issuecomment-1048490972

volconst, 2022-02-24:
Try dtr param and the many other proposals in here https://github.com/kliment/Printrun/issues/1201#issuecomment-885250173