kliment/Printrun #1250: Unable to set extrusion length

supuflounder, 2022-03-15:
For some bizarre reason it thinks that I want to extrude 1000.000000 mm of filament.  I am unable to type any value into the "length" box to change this to something sane, like 50mm.  What's wrong?  How can I fix this?  Shutting down the program and restarting it does not fix the problem.

DivingDuck, 2022-03-15:
Max extrusion length value for the field is 1000,0 mm or 1000.0 mm (depending on regional settings.

What is your Pronterface version?

Can you decrease/increase the length with the little triangle controls on the right side of the input box? See below

[image]

supuflounder, 2022-03-15:
It says PrintRun 1.6.0.  No reveal on what version of PronterFace.

 

 <https://user-images.githubusercontent.com/8143022/158484736-d191f70d-9804-4bc1-8750-82e7de265039.PNG>

DivingDuck, 2022-03-15:
Pronterface = Printrun

Had you switched between Version 1.6 and actual 2.0 RC version?

supuflounder, 2022-03-16:
I downloaded what I thought was the latest version a few weeks ago.  I’ll go back to the site later today.

supuflounder, 2022-03-16:
What little triangle?  I’ll look for a new download later today

 

 <https://user-images.githubusercontent.com/8143022/158484736-d191f70d-9804-4bc1-8750-82e7de265039.PNG>

DivingDuck, 2022-03-21:
@supuflounder 
had you solve the problem?

supuflounder, 2022-03-21:
Yes.  I got the new download.  Now I have to figure out why I am having adherence failures, but at least I can communicate to the printer reliably.

DivingDuck, 2022-03-21:
Good to know. I will close the issue.

Regarding adherence, look for the usual suspects. Clean and grease-free bed surface (do not touch the print surface with fingers after cleaning), well leveled bed, correct bed temperature for your filament, printing first layer slow and without fan or at least with little fan speed. Spread the glue with an wet paper towel thinly over the bed surface in case you use a glue stick.

Good luck and happy printing.

closed 2022-03-21 by DivingDuck

supuflounder, 2022-03-21:
I have tried using the recommended extruder and bed temperatures for PLA (215C and 60C), I clean the bed with isopropyl alcohol and do not touch it, I have used a dry dish sponge to clean it (before the alcohol swab), I have played with temperatures, first-layer-height and subsequent-layer-height.  I am using a Prusa Mk3 which does automatic bed-level compensation based on 49 test points.  It just started failing on one particular piece and I can’t get that piece to print no matter what I try.  Any good fora for discussing this problem that you would suggest?

 

Thanks

Joe

DivingDuck, 2022-03-22:
This is difficult to answer as I do not know what you are printing and your slicer settings are. Adding a brim can help the part to stick better on the surface. In addition try to increase the bed temp a little bit and use a glue stick as I mentioned before. Some use hair spray as adhesion helper. You don't need to add it for every print. It will remain its function for a couple of prints.

Regarding  isopropyl alcohol, I don't had good experiences with cleaning my print surface with it (I print mainly on glass).  I use dish washing liquid and hot water for cleaning and a good rinse with clear water afterwards.