kliment/Printrun #1251: AUR package printrun-git is broken

igoro00, 2022-03-20:
I tried to install `printrun-git` package from the AUR and it can't clone the repository. I'm getting this message:
```
The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
```

Basically github dropped support for git://github.com/foo/bar.git and now requires the use of https.

kliment, 2022-03-20:
Hey, sorry but I have no idea who built that package. It certainly wasn't me or anyone I know and we can't really do much about it.

igoro00, 2022-03-21:
You can actually. You're the maintainer of this software and both packages in the AUR are broken. Binary version is severely outdated(and because some dependencies for python2 don't exist anymore in arch repos it doesn't compile) and git version doesn't compile. 

Request the packages to be orphaned. `printrun-git` will require manual check by "Trusted Users" but tell them you are the maintainer of the original project and I'm 99% sure they'll let you take over the package.
`printrun` will be automatically orphaned since it's been flagged as outdated for a long time.

Here's how you can do it.
https://wiki.archlinux.org/title/AUR_submission_guidelines#Requests

Then adopt these packages.
Here's some tutorial on how to do it:
http://kokizzu.blogspot.com/2016/11/how-to-become-aur-package-adopter.html?m=1

igoro00, 2022-03-21:
> Sorry, no, I'm not willing to adopt a package someone else made that runs
> on a system I don't use and have no idea how to support. If you'd like to
> volunteer to adopt this package, I would very much welcome the help.

I can request the packages to be orphaned and look into fixing the
`printrun-git` package but you can't count on me that I'll be there to
support it in the future (like changing the version and the link to the
tar.gz after an update).

That's why I think you should be the owner of the package and/or at least
change the readme to say that the AUR package is not officially supported
and may break at any time. Because now it looks like the `printrun` is the
"official way" to install it on Arch Linux.

kliment, 2022-03-21:
I understand that, and that's exactly how we ended up like this - someone contributed a package and then ran away. Is there a way to find out where the original arch package came from and who made it?

rockstorm101, 2022-03-21:
Hi @ejona86, I believe you were the originator of Printruns's AUR package. Could you give us a hand here?

ejona86, 2022-03-26:
The way to handle this is to mark the package out-of-date (or even make a comment on AUR). It seems this happened as I got an AUR notification on Tuesday. I fixed it on Thursday. Everything should be good now.

The printrun package was very recently broken by [a deletion request for python2-pyglet](https://lists.archlinux.org/pipermail/aur-requests/2022-March/068505.html). I'm not sure how I'll be handling that yet. I'm looking forward to a 2.0.0 release.

rockstorm101, 2022-03-28:
Thanks for your insight @ejona86. I'll close this issue then since the problem is a packaging matter which does not strictly concern upstream.

closed 2022-03-28 by rockstorm101