kliment/Printrun #1252: Threading issues for large gcode files

matthewelmer, 2022-03-28:
Sorry about the images. I found it extremely difficult to copy the console logs. The number of layers and time estimation was always the same between the three attempts, but only the third attempt ran without errors and showed the full model in the preview. Please let me know if I can provide any more info!

Gcode is attached as a .txt (hopefully? github doesn't seem to like it.)

[image]

Exception in thread Thread-12:
Traceback (most recent call last):
  File "threading.pyc", line 808, in __bootstrap_inner
  File "threading.pyc", line 761, in run
  File "printrun\pronterface.pyc", line 1452, in loadviz
  File "printrun\gcview.pyc", line 338, in addfile_perlayer
  File "printrun\gl\libtatlin\actors.pyc", line 429, in load_data
MemoryError: cannot allocate memory for array

Loaded F:\Documents\3D Printer Files\LANL\0.6mm nozzle gcode

[image]

DivingDuck, 2022-03-29:
You can zip the files and attach the zip file here. What is your printrun version and operating system?
I just run my installation and open a file with 1.2 million lines without a problem. Same for the actual release candidate version 2.00RC8.
Please attach the gcode file if possible. I will check if I can replicate the problem.

Can you remember , what exactly you did after you start opening the file? Did you maybe try to open the gcode view while printrun was in the process loading the file?

volconst, 2022-03-30:
@matthewelmer , are you using precompiled binary or running from source.
Running from source requires compiling a native class for conserving memory

matthewelmer, 2022-04-04:
I'm pretty sure I'm using the precompiled binary since this is my windows machine. (I get that there may be ways to compile for windows, but I've never done so.) I should add that I was wiggling the graphical view before it had finished loading, and when I let it load without wiggling it there are no problems. Here's the gcode
[pistons_0_1.zip](https://github.com/kliment/Printrun/files/8406061/pistons_0_1.zip)
!

DivingDuck, 2022-04-04:
@matthewelmer, 
thank you for attaching the model. Opening your gcode in Printrun works fine for me, but there is indeed a little glitch. Not critical but it's there.

OS is Windows, Windows 10 x64 I guess? What is the version of Printrun?  :)

You need to give Printrun time to open the file completely. This is also the workaround for your problem. You will recognize the progress of loading on the red marked areas below:

[image]

As soon as loading is complete you will see that the status bar at the bottom left is empty and terminal window on the right will tell you the stats of your model as highlighted  below:
 
[image]

Please try not to view layers in the Gcode view window during the time where Printrun load the model because, as you find out, there is a not solved user action that will cause an error report. The model will nerveless load correctly in case you wait for finishing the import of the model.

DivingDuck, 2022-04-04:
@volconst, @kliment 
I took only a quick look on the source code and saw what cause the problem. In the end it seems to be an not expected user action for the time a model is loading and where a user open the Gcode layer view and try to move the layer slider and where maxLayer isn't initialized / haven't a defined value.
The model is importing correctly and after this is finish you can use the slider in the window correctly.
  
[image]

[image]

I will mark this issue as an bug so that we can fix it as soon we find some time (can't quickly see/find the point where this need to be fixed). I'm a bit in hurry for now and quite busy this week.
Best regards, 
DD

matthewelmer, 2022-04-04:
Yeah, there's no need to fix this quickly. I'll just wait till my model is done before viewing :)

Thanks all!

DivingDuck, 2023-02-12:
@matthewelmer,

it is quite some time ago but we finally got around it. 

As mentioned those errors came from a not completely loaded Gcode model. I did change the behavior for the Gcode Viewer so that is do not process the view if the slider move to an not existing position while loading the model. An additional information printed in the console window inform the user to wait until the model is loaded completely.

Here is a link to the latest dev build: https://github.com/kliment/Printrun/actions/runs/4152400998

I will close the issue and mark as completed. (https://github.com/kliment/Printrun/commit/0c296bab07a4e6a11470082156cf078f745a54a2) Thanks for your report

PR #1308 (2023-03-15): Release Printrun 2.0.0
Now that #928, #1171 have been dealt with and Windows side has been revamped (many thanks @DivingDuck) I believe we are in shape for finally releasing 2.0.0. Up for debate of course, please speak up if you know any show-stoppers.

Two things I can think of that'd be nice to do before releasing:
 - [x] Test and confirm macOS binaries work as expected
 - [x] Revisit Windows installation instructions and update (if required)

--
Fixes #1305
Fixes #1296
Fixes #1295
Fixes #1252
closed 2023-02-12 by DivingDuck