kliment/Printrun #1253: Docked 3d view is very slow

Wallby, 2022-04-04:
Pronterface version 2.0.0rc7 on LMDE 5 (based on Debian Bullseye).

Nor "use a lighter visualization" nor 2D/3D affects the framerate. However.. double clicking to open the view in a seperate window (which is seemingly always a 2D view) has no issues whatsoever (usable fps not noticibly slow in any way). Is the fps capped or something?

volconst, 2022-04-05:
Try hiding the jog controls (left pane), so that 3d view is left and log pane right.
Does the 3d view refresh?

Wallby, 2022-04-06:
@volconst  jep that works.  What also works is hiding the "command pane" (the rightmost pane) or reducing the size of the command pane. Making the command pane bigger again reintroduces the issue. If the jog controls are collapsed then the size of the command pane doesn't ever cause the issue to happen.

volconst, 2022-04-06:
@Wallby this is a bug in wxwidgets library
There is a pull request to require a newer version of wxwidgets, which is not yet committed.
https://github.com/kliment/Printrun/pull/1171#issuecomment-1068382131
To confirm, can you send this as a gcode command, no need to connect to printer, and post the printed version
```
!import wx; print(wx.__version__)
```
You can check the conversation and tell whether you think it is the same problem.

Wallby, 2022-04-07:
@volconst Yes that outputs 4.0.7 for me too. Sounds likely to be the same issue.

volconst, 2022-04-11:
@Wallby , the fix is to upgrade to wxpython>=4.1
```
pip3 install -U 'wxpython>=4.1'
```
For this to succeed you have to install various linux development packages.
What is the exact list of dependencies I do not know.
See https://www.wxpython.org/blog/2017-08-17-builds-for-linux-with-pip/index.html

closed 2022-06-02 by volconst