kliment/Printrun #1254: MacOS 12 support on M1 silicon

deveshsnair, 2022-04-17:
Hi I can't open pronterface in MacOs 12 on the M1 mac is there any solution for it I'm bit new to ARM.

rockstorm101, 2022-04-18:
Possible duplicate of #1129. CC @a2k-hanlon. I believe there's a [macOS Big Sur 11 runner][1] available now? Will it work for macOS 12? Not sure it includes the ARM architecture though.

[1]: https://github.com/actions/virtual-environments/blob/main/images/macos/macos-11-Readme.md

a2k-hanlon, 2022-04-19:
Hi! I would say this is actually a duplicate of #1191. The Mac build file permission problem in issue 1129 was fixed last year in 2.0.0rc8. As of issue 1191, the python app did not run successfully on M1 machines when packaged on an x64 machine. At the time, it was built using the macOS 10.15 runner. I can investigate to see if the new macOS 11 runner improves this situation.

In fact, the most recent Github action builds on the repo should already be using the new macOS 11 runner, since the `macos-latest` OS option specified in the [macOS build workflow file](https://github.com/kliment/Printrun/blob/master/.github/workflows/buildpackage-mac.yml) now points to macOS 11.

So @deveshsnair, could you please try this binary and let me know if it works for you? https://github.com/kliment/Printrun/suites/5809666380/artifacts/194407693

If that binary doesn't work, please try the workaround of building the app from source, as described in the README, in order to run it on an M1 Mac. If you do try building it from source, please let me know how that goes too.

rockstorm101, 2022-05-28:
Can we confirm whether the [latest binary built for macOS][1] works on an M1 Mac? Hoping to close out this issue and #1191.

CC @DumbGameMaker

[1]: https://github.com/kliment/Printrun/suites/6424706897/artifacts/234770484