kliment/Printrun #1255: prontaface bootup but is not displayed

suicideroade, 2022-04-22:
please can someone tell me why the task manager shows prontaface but the program is not visible!
I used win10 64bit on surface pro 7 tablet and in the task manager it says protaface32bit but I can't find a 64bit version
deleting the complete printrun folder and then downloading the program again didn't help either, but i used prontaface on the device mentioned above

suicideroade, 2022-04-22:
Traceback (most recent call last):
  File "pronterface.py", line 34, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "printrun\pronterface.pyc", line 32, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "printrun\pronsole.pyc", line 42, in <module>
  File "zipextimporter.pyc", line 82, in load_module
  File "printrun\power\__init__.pyc", line 20, in <module>
  File "platform.pyc", line 1265, in system
  File "platform.pyc", line 1180, in uname
  File "platform.pyc", line 504, in _syscmd_ver
  File "platform.pyc", line 444, in popen
RuntimeError: Can not locate 'w9xpopen.exe' which is needed for popen to work with your shell or platform.

volconst, 2022-04-22:
What pronterface version do you use?
Get x64 version from 
https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc8

rockstorm101, 2022-05-28:
Hi @suicideroade, did you try using the x64 version as suggested? Did that solve the issue? Thanks

suicideroade, 2022-05-29:
Hello,yes it works as far as I can judge everything perfectly. Thank you!Kind regardsRené Sec-------- Ursprüngliche Nachricht --------Von: Rock Storm ***@***.***>Datum: Sa., 28. Mai 2022, 14:09An: kliment/Printrun ***@***.***>Cc: suicideroade ***@***.***>, Mention ***@***.***>Betreff: Re: [kliment/Printrun] prontaface bootup but is not displayed (Issue #1255)
Hi @suicideroade, did you try using the x64 version as suggested? Did that solve the issue? Thanks

closed 2022-05-30 by volconst