kliment/Printrun #1256: Python Scripts Folder not on Path for MacOS

adam-grant-hendry, 2022-04-28:
When trying to use `twine` to upload a MacOS built wheel to PyPI, I get the error

```bash
WARNING: The script twine is installed in '/Library/Frameworks/Python.framework/Versions/3.8/bin' which is not on PATH.
```

This is not an issue on Windows and Ubuntu builds.

1. How can I add the folder to `$PATH`?
2. For matrix builds, is there an environment variable for Python scripts folder, or do I have to hard code this for MacOS builds?

volconst, 2022-04-28:
On a linux system:
```
export PATH=/Library/Frameworks/Python.framework/Versions/3.8/bin:$PATH
```
Verify with
```
echo $PATH
```
If some dir contains space then you have to enclose with double quotes.

rockstorm101, 2022-05-28:
Hi @adam-grant-hendry, did the proposed solution work for you? If so, would you consider this matter closed?

adam-grant-hendry, 2022-05-28:
This seems an inelegant solution to hardcode the path. If testing with tox, I have to specify for every version of python being tested.

Can an environment variable be added that works across all OS's and can be configured for matrix builds?

rockstorm101, 2022-05-31:
Please forgive my ignorance on macOS but, how is this related to Printrun? Isn't `/Library/Frameworks/Python.framework/Versions/3.8/bin` a path for the Python binary itself?