kliment/Printrun #1257: Feature Requests

dmitche3, 2022-05-04:
i'm sorry but I can't figure out how to mark this as feature requests.
The fire one is to add into the code a pause to change filament into the software.  Many printers(Ender 3 V2) don't work well and creating a macro isn't the bes solution I've found out.  If the printer would pause sending codes, retract, lift the hotend, etc, and wait for input on the keyboard before resuming it would be nice to have

Second, I've had failed prints because of tangled filament or running out. Many things that I print the height is not critical and I'd simply like to lower the hotend and continue printing.  But because the G-Code using absolute positioning I can't do that.  It would be great to have Printrun, pause the print, move the hotend to the desired location, restart the print translating the difference between the G-Code's absoulte position to the new relative position.

A third cosmetic would be to the macro display.  Adding a line separator between the rows. I keep my macros grouped by Bed Temp, HotEnd Temp, Fan Speets, Flow compensation, and Movement commands in order to avoid the accidental depressing a move command by mistake while printing.  While I a bad work-around method helps, by creating a macro with a long list of "-----" it doesn't work if the image panel is resized. 

Thank you.  And I love this program and wish that I'd found it before I did.  Now I can't live without it.

volconst, 2022-05-05:
@dmitche3 What is the problem with pause button and macro for retracting, lifting?

For the absolute coords when resuming, you can try this pull request https://github.com/kliment/Printrun/pull/1197
You have to run from source from the PR branch, because no binary is available.
The feature is not finished, but could be workable.

Can't you use https://marlinfw.org/docs/gcode/G092.html for the absolute coords.

dmitche3, 2022-05-06:
Thank you for the response.  To answer the questions to an email that I recieved from Kilment, I'm posting this for others.

I was not aware of how to use the G92 code in order to change offsets while printing.  Thank you for nudging me in that direction.

As for using the pause G-Code I've had no luck doing so.  I hadn't thought about hitting the pause button manually and using a macro at that point. (Hand slapping forhead)

Also, there appears to be a feature that is in testing #1197 that will allow to reprint at a layer height.  This sounds extremely promising for issues of filament breakage, run-outs, etc.  Whether it makes it to production I'll wait and see.

Thank you Kliment.