kliment/Printrun #1258: TypeError when adjusting temp, speed, flow rate

Swarvey, 2022-05-06:
This might not be the most common issue as I have found zero information everywhere else. Problem is as follows:

while clicking on / entering temperatures, flow rates, print speed etc, the log shows an error, referencing controls.py line 234

"TypeError: Slider.SetValue(): argument 1 has unexpected type 'float' "

Under Ubuntu 21, when this error showed up, the application would continue functioning and set the requested temp, speed, flow rate etc. Updated to Ubuntu 22 yesterday and now once the error message shows in the log, Printrun freezes and cannot be interacted with until either killed, or the print finishes.

Further information, after updating to Ubuntu 22, I had to rebuild and reinstall Printrun to get it operational again, as well as install/update to Python3.

To quote the Bloodhound Gang, I'll appreciate your input

volconst, 2022-05-08:
@Swarvey , please test if PR #1252 fixes the error
Wheels artifacts can be found at https://github.com/kliment/Printrun/actions/runs/2285967572#artifacts
or checkout and run from source branch https://github.com/volconst/Printrun/tree/dbl-spinner

Swarvey, 2022-05-09:
I can confirm this fixes the error in the log screen.

Printrun is still freezing on Ubuntu 22 but I can chalk that up to in-place-upgrading the OS

PR #1262 (2022-05-09): Fix TypeError when adjusting speed, flow rate
Fixes #1258
commit 302ed8a1 (2022-05-09): Fix TypeError when adjusting speed, flow rate

Fixes #1258
closed 2022-05-09 by kliment